dart-archive/sdk #7: void/unit types in IDL definitions

zerny, 2015-02-16:
Definitions would read better if we had short-hands for definitions that don't require content:

For methods:

  service Foo { void bar(...); }

For structs:

  struct SExp { union { unit nil; Cons cons; } }
  struct Cons { ... }

or just omit the type:

  struct SExp { union { nil; Cons cons; } }

or we keep spreading the not-so-well chosen 'void' :-)

zerny, 2015-02-17:
A small comment regarding style. When building unions the use of setters for creating/tagging non-struct cases is opaque. Also, using setters can't be made consistent for nullary constructors.

```
struct Node {
  nil;
  int32 num;
  Cons cons;
}
```

building would likely be something like:

```
NodeBuilder node = ...;
if (...) node.nil();
else if (...) node.num = 42;
else if (...) node.initCons();
```

It would be more clear if the methods for selecting the class/tag of node gave an indication of this like it is done for the struct case with the 'init' prefix. We could use init in all cases, albeit with a void return for the non-struct cases. Or maybe 'selectX' since we really are committing to the class/tag for node.

Naturally, if we have initNil() and initNum(n) it would be nice to have initCons(fst, snd) but then we need support for bottom-up construction too.

kasperl, 2015-02-17:
I agree that we should look at the style. First CL (https://codereview.chromium.org/933803002/) for introducing support for void methods and union slots doesn't really deal nicely with this and the names chosen for setting a 'void' union slot is `node.setNil()` in both Dart and C++.

kasperl, 2015-02-17:
Fixed in commit e479ea4d937512b87558f2483ae8cb1a66ddafae. Will open a new issue for the style issues.

closed 2015-02-17 by kasperl