dart-archive/sdk #8: Instantiated return types in IDL definitions.

zerny, 2015-02-16:
Definitions with a service of List<T> return type fail to compile:

> service Foo { List<int32>* bar(); }

fails with:

> Unhandled exception:
> end of input expected at 1:1
> #0      Failure.value (package:petitparser/src/core/context.dart:107:16)
> #1      parseUnit (package:servicec/src/parser.dart:14:30)
> #2      compile (package:servicec/compiler.dart:23:24)

kasperl, 2015-02-17:
I don't think we support returning List* (should we) but we'd like to support returning List<T> where T is a struct type (by value) or a primitive type. Perhaps, we should also support List<T*> where T is a struct type, but I don't think we support that in general.

kasperl, 2015-02-17:
This is really a request for allowing methods to return and accept a single `List<T>` argument, where T can be a struct type or a primitive type.

closed 2016-05-19 by mit-mit