dart-archive/sdk #12: Add Process Spawning Support

kaendfinger, 2015-03-01:
I'm planning to use fletch to run my new OSX API I am writing, because I want it to be synchronous. I would like to request that the ability to spawn external processes with fletch. This will complete all the things I need to move to using fletch.

kasperl, 2015-03-01:
Is this something you build on top of dart:ffi? If you can cook up a patch,
we'd love to see a pull request!

kaendfinger, 2015-03-01:
@kasperl my OSX api is not, at the moment, it's running external processes. I could look into building an external process executor on FFI, but I do plan on interop between Obj-C and Dart.

kasperl, 2015-03-02:
I guess my point was that they way all access to OS level resources is implemented at the moment is through dart:ffi, so the external process execution support should also be built that way. How much do you need and would you care to propose an API (preferably synchronous)?

kaendfinger, 2015-03-02:
@kasperl 
I need the following:

- Ability to control the process working directory
- Ability to read and write to the process through stdin/stdout/stderr
- Ability to get the exit code

```dart
var process = ExternalProcess.start("osascript", ["-ss", "-e", 'tell "Google Chrome" to activate']);
var out = process.stdout.read();
var err = process.stderr.read();
process.stdin.write(bytes);
var exitCode = process.waitFor();
```

I would use threads to handle the I/O in a workable way.

skabet, 2015-03-02:
I've started some initial design for a Process API, see https://codereview.chromium.org/969733002.

The plan is to start out with the bare minimum, to run/wait processes, and expand into stdout/err/io communication. As you write yourself, threads seams to be an obvious way to handle this.

Also, I'm intrigued by the name ExternalProcess :)

kaendfinger, 2015-03-02:
@skabet Process is taken afaik :P It was either than or `SystemProcess`