dart-archive/sdk #14: Can't add Fletch to path

Scorpiion, 2015-03-09:
If you build Fletch, and then tries to add the path of one of the output directories to your own PATH, fletch fails to execute.

Sample output that fails:
```bash
$export PATH=${USER}/Workspace/fletch-repo/fletch/out/ReleaseX64/:$PATH
$ fletch hello.dart 
ERROR: Cannot open ../../lib/core/core.dart
Failed to read from socket
../../src/vm/session.cc:82: error: Compiler crashed. So do we.
Aborted (core dumped)
```

Sample output that works:
```bash
$cd /home/robert/Workspace/fletch-repo/fletch/out/ReleaseX64/
$./fletch hello.dart
```

Sample output that works (my current workaround):
```bash
$alias fletch="${USER}/Workspace/fletch-repo/fletch/out/ReleaseX64/fletch"
$./fletch hello.dart
```

Created after conversation on gplus: https://plus.google.com/+PetervonderAh%C3%A9/posts/dheqctsNvHn

peter-ahe-google, 2015-03-09:
The driver I'm working on can correctly identify the location of a symlinked executable. So hopefully this should get fixed real soon.

kasperl, 2015-05-04:
The driver now works well enough that you can try it out:

`out/ReleaseX64/fletch_driver hello.dart`

You may be able to put that into your path.

Scorpiion, 2015-05-05:
Hi, I synced up my sources now (with gclient sync) and recompiled fletch, but now I can't run it at all. Not sure if I have missed something.

I get:
```bash
robert@DV2:~/Workspace/fletch-repo/fletch$ ./out/ReleaseX64/fletch hello.dart 
ERROR: Cannot open ./out/ReleaseX64/../../lib/core/core.dart
Failed to read from socket
../../src/vm/session.cc:89: error: Compiler crashed. So do we.
Aborted (core dumped)
```

And, yeah, that file does not exists:
```bash
robert@DV2:~/Workspace/fletch-repo/fletch$ ls lib/core/
core_patch.dart
```
(it does not exists here in the repo either: https://github.com/dart-lang/fletch/tree/master/lib/core)

Should it have been added during the build process?

madsager, 2015-05-05:
This is all a bit confusing. We changed the way to run fletch. You should no longer use 'fletch' you should use 'fletch_driver':

out/ReleaseX64/fletch_driver hello.dart

Hope that helps. :)

Scorpiion, 2015-05-05:
Ah, thanks @madsager. Now I can run fletch as before. On the subject of adding fletch to the OS path, that does still not seem to work. 

This works:
```bash
robert@DV2:~/Workspace/fletch-repo/fletch$ ./out/ReleaseX64/fletch_driver hello.dart 
Using config file: /home/robert/Workspace/fletch-repo/fletch/.fletch
Hello!
```

This works:
```bash
robert@DV2:~/Workspace/fletch-repo/fletch$ /home/robert/Workspace/fletch-repo/fletch/out/ReleaseX64/fletch_driver hello.dart 
Using config file: /home/robert/Workspace/fletch-repo/fletch/.fletch
Hello!
```

This does not work ("/home/robert/Workspace/fletch-repo/fletch/out/ReleaseX64" is part of PATH)
```bash
robert@DV2:~/Workspace/fletch-repo/fletch$ fletch_driver hello.dart 
Using config file: /home/robert/Workspace/fletch-repo/fletch/.fletch
Invalid argument(s): [fletchVm]: No fletch VM at 'file:///home/robert/Workspace/fletch-repo/fletch/fletch'.
```

This does not work either, this worked before (see first comment in this issue):
```bash
robert@DV2:~/Workspace/fletch-repo/fletch/out/ReleaseX64$ ./fletch_driver hello.dart 
Using config file: /home/robert/Workspace/fletch-repo/fletch/.fletch
Invalid argument(s): [fletchVm]: No fletch VM at 'file:///home/robert/Workspace/fletch-repo/fletch/fletch'.
```

peter-ahe-google, 2015-09-15:
Suggested fix: https://codereview.chromium.org/1347463003

peter-ahe-google, 2015-09-16:
Fixed in 039f4c4a73130ecf2cbb947b2b58e9bff7a9b44e

closed 2015-09-16 by peter-ahe-google