dart-archive/sdk #16: Access to Port objects is guarded by a SpinLock: This can potentially cause a lot of spinning

mkustermann, 2015-04-15:
To synchronize access to Port objects, a spinlock is currently used. Since the kernel can interrupt a native thread at any point in time (even if the lock is held), it might cause other native threads to spin for significant amounts of time before they can acquire the lock. If the system is loaded for some period of time and not all native threads can run in parallel, a suspended native thread which holds a lock can cause a lot of wasted CPU cycles.

Possible things to look at:
- Clearly document this behavior on the Lock/Unlock methods of the Port class
- Ensure we hold on to the lock for as little time as possible, never do syscalls while holding the lock (e.g. no memory allocation), ...
- Avoid locking the port and assuming another component in the system will free it within reasonable amount of time (this is very hard to reason about, and evolving code might make the time longer)
- Make benchmarks which try to cause contention on the lock to test it in real life
- Try to compare using pthread mutex, or the lower-level futexes on linux, see why there are so much slower (they should operate in user-space in non-contended case as well)

skabet, 2015-10-28:
I don't believe this is a valid bug - as in something we plan on "fixing". The entire Scheduler (and messaging) is based around lock-free structures, atomics and spin-locks. If we experience performance issues at some point, that appears to be related to this (not seen yet), we could reopen this one.

mit-mit, 2015-12-07:
@mkustermann is this relevant for FixIt week?

mkustermann, 2015-12-07:
I don't think it is. The bug report is about bad/worst case behavior of the way we do locking in a number of places (this is independend of lock-free datastructures and atomics - just about spinlocks).

There is a way to mitigate the issue, but before changing our implementation I think it would make sense to measure it first.

skabet, 2015-12-07:
What abort reopening the issue when we actually have some data. All data
collected so far indicates that there is not an issue. I'm of the opinion
that bug trackers are for validated issues, not speculations. If it were, I
could easily file 10 bugs.

sgjesse, 2015-12-08:
Closing this. We can look into this if it starts showing up on profiles.

closed 2015-12-08 by sgjesse