dart-archive/sdk #17: tools/library_combiner.py fail in ubuntu 14.04 x64

DanCraft99, 2015-04-20:
tools/library_combiner.py fail in ubuntu 14.04 x64 with the line 
..
os.system('echo -e "%s" | ar -M' % script)

by creating a temporary file, the following code works:

def invoke_ar(args):
  library_name = args[len(args) - 1]
  libraries = args[:-1]
  mri_file = open("combiner.mri", "w")
  mri_file.write('create %s\n' % library_name)
  for lib in libraries:
    mri_file.write('addlib %s\n' % lib)
  mri_file.write('save\n')
  mri_file.write('end\n')
  mri_file.close()
  #os.system('echo -e "%s" | ar -M' % script)
  os.system('ar -M < combiner.mri')

madsager, 2015-04-20:
Dan, thanks for the report! Could you post the output of 'ar --version' on your system. I have tested locally with version 2.24 and 2.25 and they both work with the current combiner script. I believe I have a 14.04 system at home, so I'll have a look from home later if I cannot reproduce it here.

madsager, 2015-04-20:
OK, never mind. This is not 'ar' it is python os.system that cannot find echo. I managed to reproduce and using a full path for 'echo' makes it work. Will fix.

madsager, 2015-04-20:
Thanks again for the report. I believe the following change fixes the issue:

https://codereview.chromium.org/1089683004/

Please reopen this issue if it still doesn't work.

closed 2015-04-20 by madsager