dart-archive/sdk #19: Issues with tearoffs

zerny, 2015-04-22:
Currently tearoffs are not considered functions and can not be called without disambiguating the receiver at the call site:

```dart
class Foo {
  Function fn;
  Foo() {
    fn = () => 42;
  }
}

main() {
  Foo foo = new Foo();
  print(foo.fn is Function); // should print true, currently prints false.
  print((foo.fn)()); // prints 42.
  print(foo.fn()); // results in an error.
}
```

skabet, 2015-04-23:
Proposed fix: https://codereview.chromium.org/1103663002/

kasperl, 2015-04-23:
@skabet The patch only solves the 'is Function' part right?

skabet, 2015-04-23:
Ah yes, I meant to split this up. There are two issues here.

skabet, 2015-04-23:
See #20 for the closure in field problem.

skabet, 2015-04-24:
Closing as eb63ad274f0d19aa953d9426dec4e816ec439557 landed, and #20 is tracing the field problem.

closed 2015-04-24 by skabet