dart-archive/sdk #23: VM Exception when load simple dart script

DanCraft99, 2015-05-01:
check out latest code, make
fletch hello world.dart 

Exception: process not implemented.

Mac 10.10.1
Ubuntu 14.04 64
Fletch 32/64 bit both failed.

Can give update? Eg the old fletchc deprecated. And new dart2js byte ode generator is not ready. 
Some kind of status update is so helpful to outsiders.

sandrobilbeisi, 2015-05-04:
I built fletch and ran the tests as described in the wiki https://github.com/dart-lang/fletch/wiki/Building
the tests run fine, but nothing else actually works
not even hello.dart
```dart
void main() {
  print("Hello, World!");
}
```

I launch it as /DevExtra/fletch-repo/fletch/out/ReleaseIA32Clang/fletch /DevExtra/fletch-repo/fletch/hello.dart 
all I ever get is

>execvp failed: No such file or directory
>../../src/shared/native_process_posix.cc:51: error: unimplemented code
>execvp failed: Undefined error: 0
>../../src/shared/native_process_posix.cc:59: error: unimplemented code

Mac OS X 10.9.5

kasperl, 2015-05-04:
Sorry. We're changing how you invoke `fletch` quite a lot. I've updated the wiki page to reflect this, but the short answer is that you'll need to run:

`
  out/ReleaseIA32Clang/fletch_driver hello.dart
`

rather than go direclty through `out/ReleaseIA32Clang/fletch`. We expect to change this again soon.

sandrobilbeisi, 2015-05-04:
WOW ! thanks for the *very* prompt response !

closed 2015-05-21 by skabet