dart-archive/sdk #24: Sample demo code to show how callback mechanism in fletch.

DanCraft99, 2015-05-01:
there are two new ways of interfacing mechanisms to third party layers, the first is ffi and the other is idl interface bindings. Please help provide sample code on how to implement callback for the fowling cases

1) ffi
Fletch calls the third party c api, and register a callback function for third party api to callback.

2) idl interface service
The third party provide a callback function to fletch, and fletch calls the third party callback.

If two way event handling are possible for both ffi and idl service, that is good with an example code.
Thanks.

kasperl, 2015-05-04:
I will try to write up a few samples. Thanks for reporting / requesting this!

peter-ahe-google, 2015-09-07:
@kasperl ping :-)

mit-mit, 2015-12-07:
We don't currently support call backs. We opened #351 351 to track that.

closed 2015-12-07 by mit-mit