dart-archive/sdk #26: Crashing on Ubuntu for me

jpedrosa, 2015-05-11:
Hi,

I built an updated version today and I noticed that it was crashing for me. I tried to track it down when it started by bisecting the git commits and I think it started with the following commit:

6eb6ecd Prepare for supporting the three variants of InvokeTest. BAD - CULPRIT

Here's some sample output:

$ /home/dewd/dice/somefletch/fletch/out/ReleaseIA32/fletch_driver hellow.dart 
Using config file: /home/dewd/repo/github/arpoador/direct_test/momentum/.fletch
../../src/vm/session.cc:558: error: unreachable code
Non-zero exit code from '/home/dewd/dice/somefletch/fletch/out/ReleaseIA32/fletch' (-6).
Abortado (imagem do núcleo gravada)
$ cat hellow.dart 
main() {
print("Hello");
}

It doesn't always produce the same error. Sometimes it just hangs.

Platform is Ubuntu 14.10 64 bits.

kasperl, 2015-05-12:
You may have been bitten by the fact the the Fletch compiler is a long running process that stops for no-one. Try killing all Dart processes on your machine (or find the right one and just kill that) and try again.

    $ killall -v dart
    $ fletch_driver hellow.dart

jpedrosa, 2015-05-12:
Thanks. I think that was the case. Here's what I had running:

$ ps auxf | grep fletch
dewd     27355  0.0  0.0  16668  2036 pts/1    S+   04:58   0:00          |   |   \_ grep --color=auto fletch
dewd      4544  1.6  3.1 1993012 258872 ?      Ssl  Mai11  16:05          \_ /home/dewd/dice/yofletch/fletch/out/ReleaseIA32/dart -c -p /home/dewd/dice/yofletch/fletch/package/ package:fletchc/src/driver/driver_main.dart /home/dewd/repo/github/arpoador/direct_test/header_parser_kickstart/.fletch
dewd      4655  1.6  3.0 753944 249164 ?       Ssl  Mai11  15:55          \_ /home/dewd/dice/yofletch/fletch/out/ReleaseIA32/dart -c -p /home/dewd/dice/yofletch/fletch/package/ package:fletchc/src/driver/driver_main.dart /home/dewd/repo/github/arpoador/direct_test/momentum/.fletch
dewd      5720  1.6  3.4 829500 280224 ?       Ssl  Mai11  14:51          \_ /home/dewd/dice/yofletch/fletch/out/ReleaseIA32/dart -c -p /home/dewd/dice/yofletch/fletch/package/ package:fletchc/src/driver/driver_main.dart /home/dewd/repo/github/arpoador/direct_test/momentum/.fletch
dewd     10288  5.4 15.0 1856768 1226152 ?     Ssl  Mai11  40:17          \_ out/DebugIA32/dart -c -p /home/dewd/dice/somefletch/fletch/package/ package:fletchc/src/driver/driver_main.dart /home/dewd/dice/somefletch/fletch/.fletch

:-)

Now it is working again after I cleared those processes.