dart-archive/sdk #27: File#write is not setting permissions on newly created file.

jpedrosa, 2015-05-13:
Run this code:

import "dart:io";
import 'dart:typed_data';

codeUnitsToUint8List(s) {
  var len = s.length, i, a = new Uint8List(len);
  for (i = 0; i < len; i++) {
    a[i] = s.codeUnitAt(i);
  }
  return a;
}

main() {
  var f = new File.open("dest.txt", mode: File.WRITE);
  f.write(codeUnitsToUint8List("foo\nbar").buffer);
  f.close();
}

When running it, I'm getting this on Ubuntu 64 bits:

$ /home/dewd/dice/kingfletch/fletch/out/ReleaseIA32/fletch_driver test_writing.dart 
Using config file: /home/dewd/repo/github/arpoador/direct_test/io_caramel/.fletch
$ ls -l dest.txt 
---------- 1 dewd dewd 7 Mai 13 18:41 dest.txt
$ chmod +r dest.txt 
$ cat dest.txt 
foo
bar

I expected it to create the file with default user permissions to read and write. :-)

jpedrosa, 2015-05-15:
Hi,

As a follow up, I figured out that Fletch didn't mean to have that support for files yet.

Rather than wait, I tried to add that support myself here: https://github.com/jpedrosa/arpoador/blob/master/lib/sys/moresys.dart

I started a class called MoreSys that uses Fletch's ffi support. I've found Fletch's ffi support to be super fun! I hope others will eventually give it a try as well.

But I think we'll need the Fletch's project guidance for better cross-platform support. :-)

Thanks!

skabet, 2015-05-20:
Ah, yes, I did not consider file permissions. Will fix!

Thanks,

- Anders

mit-mit, 2015-12-07:
@skabet is this still an issue? Tagging as FixIt candidate.

ricowind, 2015-12-09:
On my machine, a slightly updated version of this works fine, it will set the default file permission:
```
import 'dart:typed_data';
import 'package:file/file.dart';

codeUnitsToUint8List(s) {
  var len = s.length, i, a = new Uint8List(len);
  for (i = 0; i < len; i++) {
    a[i] = s.codeUnitAt(i);
  }
  return a;
}

main() {
  var f = new File.open("dest.txt", mode: File.WRITE);
  f.write(codeUnitsToUint8List("foo\nbar").buffer);
  f.close();
}
```

closed 2015-12-09 by ricowind