dart-archive/sdk #28: Feature request: stdout.write or similar that didn't include the newline character by default.

jpedrosa, 2015-05-19:
Hi,

While playing with Fletch one of the things that I wanted to be able to do was to print to the terminal to create a kind of progress bar or count down. Sort of like running this code in the terminal:

    while true; do echo -ne "`date`\r"; done

In one of the Python scripts I noticed they were doing something similar so I wanted to do it with Fletch as an experiment and to learn how to do it. All we needed was to be able to print to stdout without including a newline by default, but we should include the carriage return (\r) instead and then by the next print we can override the content and create some kind of progress bar or count down.

I tried to extend Fletch by using the printf function from FFI but it didn't work. Fletch seems to include its own stdout buffer which is only flushed by calls to the "print" function.

I know that the standard library is still a work-in-progress and I don't want to rush things. But I figure it couldn't hurt to ask for a potentially useful feature. :-)

Thanks!

Cheers,
Joao

seaneagan, 2015-05-19:
`stdout.write` already doesn't append `\n`, only `stdout.writeln` does that.  Am I missing something?

jpedrosa, 2015-05-19:
It's not implemented in Fletch yet, I think. I just ran this code and it didn't work with the Fletch VM:

    import "dart:io";
    main() {
      stdout.write("hey");
    }

While we wait a decision about how the standard libraries should be implemented and on how they are going to be different from the standard libraries that accompany the default Dart VM, this is potentially a useful API on its own that could be worth it having it implemented in some form.

I'm experimenting with CGI in Fletch and I'm not sure about whether I would need a "stdout.write" function or not for that. :-)

skabet, 2015-05-20:
Hey Joao,

It's true that stdout is intercepted by the compiler on usual calls. There is however two way to get around it:

1) Use --out <file.snapshot> and give the snapshot to the fletch vm. That way there will not be any interception.
2) Call fflush(stdout) through dart:ffi.

I'll leave this bug open as it is a very valid feature request! :)

Cheers,

- ANders

jpedrosa, 2015-05-21:
Thanks for the hint! The first time I tried it, I thought I would need the correct file stream as parameter to fflush. But if the parameter is "NULL" it flushes all the streams so it works for now. :-)

I added the implementation details to my [own project](https://github.com/jpedrosa/arpoador/commit/fa5cae343121cb57b6671638642a51d3e22fea45) if anyone needs something similar for now.

kasperl, 2015-05-27:
@jpedrosa If you feel like contributing your implementation to the main Fletch repo, we'd love to see a simple PR :)

mit-mit, 2015-12-07:
`File.open("/dev/stdout", mode: File.WRITE)` ought to work, but it gives an open error. Tagging for fixit week.

skabet, 2015-12-08:
This works for me:

import 'dart:convert';
import 'dart:typed_data';
import 'package:file/file.dart';

void main() {
  var file = new File.open("/dev/stdout", mode: File.WRITE);
  file.write(new Uint8List.fromList(UTF8.encode("lalalala\n")).buffer);
}

ricowind, 2015-12-08:
I can validate that what Anders did is working, but we should probably have a stdout class with a write method and a stdin class with a read method as a convenience function in one of our packages

mit-mit, 2015-12-08:
Yes, that would be great.

skabet, 2015-12-08:
In extend to that, we need to decide how that should work with our print
interceptors. First thought is it should not be part of it.

commit 033c3054 (2015-12-10): Add getters for stdin, stdout and stderr on File

Closes #28

BUG= https://github.com/dart-lang/fletch/issues/28

R=karlklose@google.com

XXX

Review URL: https://codereview.chromium.org/1519433003 .
closed 2015-12-10 by sgjesse