dart-archive/sdk #32: String.Split isn't implemented

lukechurch, 2015-05-24:
And is pretty important for any string processing. I suspect it isn't implemented because of a dependency on regex.

I have built a dumb work around, lets decide what the strategy is here.

skabet, 2015-05-26:
Would a implementation only accepting Strings as pattern be okay for now?

madsager, 2015-05-26:
An implementation only accepting String patterns would be a great start. Most of my usages of String.split are with String patterns.

(Sorry about the accidental close and reopen).

lukechurch, 2015-05-26:
@skabet Yes that would be a great start, split by string patterns would cover almost all of my usages.

skabet, 2015-05-26:
Landed in 0fce100. Let's open a new CL for overall RegExp support.

closed 2015-05-26 by skabet