dart-archive/sdk #39: Sometimes the CPU gets pegged at 100% for longer than expected.

jpedrosa, 2015-05-31:
Hi,

While developing a [Postgres client test](https://github.com/jpedrosa/arpoador/blob/master/direct_test/postgres_line/startup_request.dart) that uses socket to connect to the backend and then needs to process the data, I noticed that sometimes the script takes much longer to finish running. When it should take about 2 seconds, it may take over 25 seconds to finish running when that happens. And while it takes longer to finish it always finishes. :-)

I can reproduce it when running [that script](https://github.com/jpedrosa/arpoador/blob/master/direct_test/postgres_line/startup_request.dart) with genSampleQuery70 or genSampleQuery80 inputs. It may take many tries to reproduce it. The issue shows up whether I'm using fletch_driver or fletch to run the snapshot. It seems to show up more often when I'm running with the fletch_driver though.

To try to find out more about the issue, I started running it with the DebugIA32 version (I usually run with ReleaseIA32.) Then when it happened again I used the gdb to capture the backtrace with "sudo gdb -p processId" and then on the gdb prompt I saw this:

    (gdb) bt
    #0  0xf77a3cb0 in __kernel_vsyscall ()
    #1  0xf776f076 in pthread_cond_timedwait@@GLIBC_2.3.2 () from /lib/i386-linux-gnu/libpthread.so.0
    #2  0x080c1873 in fletch::PosixMonitor::WaitUntil (this=0x90de0a0, microseconds_since_epoch=1433060582139310) at ../../src/vm/platform_posix.cc:215
    #3  0x080d91a3 in fletch::Scheduler::Run (this=0xffdf40e0) at ../../src/vm/scheduler.cc:232
    #4  0x080cce86 in fletch::Program::ProcessRun (this=0x90da000, process=0x9132000) at ../../src/vm/program.cc:1421
    #5  0x080ab99f in fletch::RunSnapshot (bytes=...) at ../../src/vm/fletch_api_impl.cc:31
    #6  0x080abc51 in FletchRunSnapshot (snapshot=0x90c8000 "\276", <incomplete sequence \357>, length=68685) at ../../src/vm/fletch_api_impl.cc:71
    #7  0x080a3367 in fletch::Main (argc=2, argv=0xffdf42c4) at ../../src/vm/main.cc:108
    #8  0x080a3420 in main (argc=2, argv=0xffdf42c4) at ../../src/vm/main.cc:135
    (gdb)

Twice I've captured that same bt output.

Running it with DebugIA32:

    $ time /home/dewd/dice/zenfletch/fletch/out/DebugIA32/fletch startup_request.snapshot > /dev/null
    real	0m22.117s
    user	0m22.104s
    sys	0m0.012s
    $ time /home/dewd/dice/zenfletch/fletch/out/DebugIA32/fletch startup_request.snapshot > /dev/null
    real	0m44.518s
    user	0m44.509s
    sys	0m0.016s

And with ReleaseIA32:

    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch startup_request.snapshot > /dev/null
    real	0m1.945s
    user	0m1.939s
    sys	0m0.004s
    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch startup_request.snapshot > /dev/null
    real	0m20.976s
    user	0m20.932s
    sys	0m0.040s

Generally I don't send the output to /dev/null which means that I may get more output text on the window and it's slightly slower then.

Tough issue, I know. :-)

Cheers,
Joao

skabet, 2015-06-01:
Hi Joao,

Thank you for the bug report! It sounds like we have a bug in the scheduler!

I'm not able to run your example, as I don't know how to set up a postgresql user without a password. Could you send me some instructions on how to either:

1) Set up postgresql with the needed user/database/content.
2) Find a reproducible without postgresql.

Thank you!,

- Anders

jpedrosa, 2015-06-01:
Hi,

I recently used these instructions to install Postgres on my machine:
http://www.postgresql.org/docs/current/static/install-short.html

Then I customized it a bit by creating another database called "devel":

    /usr/local/pgsql/bin/createdb devel

The table of the sample (pg_description) is actually present in every Postgres installation since it's one of the system tables used by Postgres itself. We can query the system tables to get more meta information. In this case the table had over 3,600 rows which caused more stress to Fletch and my Postgres client code, so that it triggered this issue.

I can give "2" a try. :-)

Thanks!

Cheers,
Joao

jpedrosa, 2015-06-01:
Hi,

I created a test that read the bytes directly from files rather than from Postgres sockets. The bug is not present when reading the bytes from the files.

    $ ls -la
    total 256
    drwxrwxr-x 2 dewd dewd   4096 Jun  1 15:58 .
    drwxrwxr-x 8 dewd dewd   4096 Mai 27 13:03 ..
    -rw-rw-r-- 1 dewd dewd    663 Mai 31 20:05 a_table_for_two.dart
    -rw-rw-r-- 1 dewd dewd  17467 Jun  1 16:30 buggy_wheels.dart
    -rw------- 1 dewd dewd     31 Mai 31 06:19 .fletch
    -rw-rw-r-- 1 dewd dewd    617 Jun  1 16:30 peek_bytes.dart
    -rw-rw-r-- 1 dewd dewd 209247 Jun  1 15:59 query_response_bytes
    -rw-rw-r-- 1 dewd dewd    326 Jun  1 15:11 start_up_bytes
    -rw-rw-r-- 1 dewd dewd   1376 Mai 31 18:05 startup_request.dart
    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch_driver buggy_wheels.dart > /dev/null
    real	0m2.217s
    user	0m0.000s
    sys	0m0.002s

I could now try it with writing a test server socket to connect to and read the bytes from.

I also just added the relevant files to my repository if you want to have a take on that:
https://github.com/jpedrosa/arpoador/commit/cfc6f4a26d24f7d5edebf2b78362992430d3d814

jpedrosa, 2015-06-02:
Hi,

I added the [socket server tests.](https://github.com/jpedrosa/arpoador/commit/0a7274e09088f97769072e4d948eaac5b8258215) You can run them like this:

    $ /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch_driver straw_server.dart

That will start the socket server that will serve the start_up_bytes and query_response_bytes and nothing more. It is enough to mock the data the Postgresql server sends to the original script.

I have a modified socket client script that connects directly to this straw server and you can start it like this:

    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch_driver take_a_sip.dart > /dev/null
    real	0m4.498s
    user	0m0.000s
    sys	0m0.002s

That was the fast time. When it's slow it's over 50 seconds now:

    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch_driver take_a_sip.dart > /dev/null
    real	0m49.389s
    user	0m0.003s
    sys	0m0.000s

Now the Fletch runtime is actually running into the slow path more often than when running it with the fletch_driver:

    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch take_a_sip.snapshot > /dev/null
    real	0m55.963s
    user	0m55.435s
    sys	0m0.200s

To have a barebones client I also created this other script that connects to the server and just checks that the messages have the expected lengths and counts the number of packets used to send the larger query_response_bytes:

    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch_driver check_straws_data.dart 
    Using config file: /home/dewd/repo/github/arpoador/direct_test/postgres_line/.fletch
    ["start up length 326", true]
    "Query response packet size: 44800"
    "Query response packet size: 164447"
    ["Query response length 209247", true]
    "Query response packets count: 2"
    real	0m0.405s
    user	0m0.000s
    sys	0m0.002s

The problem seems to correlate with the amount of input data that Fletch and my script have to handle in a single packet. The Postgresql server seems to send packets of 8192 bytes in size. But the Fletch socket server can send much larger packets by default.

jpedrosa, 2015-06-02:
Hi,

I've just [added](https://github.com/jpedrosa/arpoador/commit/a9acb0f09b9dc1900dcbd5be8db4d6e80ef4d41f) a packet throttling to the Straw Server test. With this change the clients can handle the incoming data without the issue:

    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch_driver take_a_sip.dart > /dev/null
    real	0m2.431s
    user	0m0.000s
    sys	0m0.002s
    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch take_a_sip.snapshot > /dev/null
    real	0m2.014s
    user	0m1.983s
    sys	0m0.008s
    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch_driver check_straws_data.dart 
    Using config file: /home/dewd/repo/github/arpoador/direct_test/postgres_line/.fletch
    ["start up length 326", true]
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 24576"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 8192"
    "Query response packet size: 24576"
    "Query response packet size: 12639"
    ["Query response length 209247", true]
    "Query response packets count: 21"
    real	0m0.651s
    user	0m0.000s
    sys	0m0.003s

It's tricky though since I added a [sleep call to the Straw Server](https://github.com/jpedrosa/arpoador/blob/master/direct_test/postgres_line/straw_server.dart) so that the packets would be sent more individually.

Also, I just ran some more tests and it's still slow at times:

    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch_driver take_a_sip.dart > /dev/null
    real	0m12.954s
    user	0m0.000s
    sys	0m0.003s
    $ time /home/dewd/dice/zenfletch/fletch/out/ReleaseIA32/fletch take_a_sip.snapshot > /dev/null
    real	0m12.692s
    user	0m12.656s
    sys	0m0.040s

I hope this has been useful to you guys. :)

I actually had some fun writing these tests. But the actual bug is still annoying.

Thanks!

Cheers,
Joao

skabet, 2015-06-02:
Hi Joao,

I've tried out your code and I was able to reproduce! Thank you so much for reducing this!

So, it turn you are right in your analysis. The bottleneck is the following:

      p(["query results", b.asUint8List()]);

in lib/postgres/client.dart.

If multiple TCP packets are sent to a host, the TCP protocol allows the packages to be collapsed into a single packet. So when you print the normal packet (~32k) it takes around 1 sec to inspect it. But sometimes they are concatenated into ~90k, which gives up to 10 sec inspect time. The reason for this is our StringBuffer implementation which is O(n^2) atm. I'll go ahead and fix that!

Thanks!,

- Anders

skabet, 2015-06-02:
StringBuffers are now much faster for your use-case than before (O(n)). This should fix the issue :)

closed 2015-06-02 by skabet