dart-archive/sdk #41: Some incremental test fails in debug_x64_asan

peter-ahe-google, 2015-06-01:
incremental/encoded/add_instance_field: Fail
incremental/encoded/instance_field_end: Fail
incremental/encoded/instance_field_middle: Fail
incremental/encoded/remove_instance_field: Fail
incremental/encoded/subclass_schema_1: Fail
incremental/encoded/subclass_schema_2: Fail
incremental/encoded/super_schema: Fail

Symptom is the same: async exception reported in dart code: socket is closed by peer.

madsager, 2015-06-01:
The VM is crashing on these tests in debug x64 asan. I'll dig into it.

madsager, 2015-06-01:
This is really weird. It doesn't happen with debug x64 clang asan; only with gcc. The crash happens during object transformation when changing method table. Tracing the transformations it is hard to see anything different in the two builds. Except that in the gcc build we all of a sudden find a garbage pointer in a heap object. I'll investigate more tomorrow, but this is pretty funky. :-\

madsager, 2015-06-02:
Turned out to be raw function pointers that turned out to be heap object tagged that we stored in the heap. Fixed in: https://codereview.chromium.org/1153733004/

closed 2015-06-02 by madsager