dart-archive/sdk #42: Observed flaky tests

peter-ahe-google, 2015-06-02:
This command may fail: ./tools/test.py -j2 -rfletchvm -cfletchc

Probably not the tests below that are flaky, more likely, the persistent process crashed.

FAILED: fletchc-fletchvm debug_ia32_clang language/malformed_inheritance_test/08
Expected: Pass 
Actual: Crash
Compile-time error expected.
CommandOutput[vm]:

stdout:
Using config file: /Users/ahe/Dart/fletch-repo/fletch/.fletch

Command[vm]: out/DebugIA32Clang/fletch -o /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_language_malformed_inheritance_test_08/fletch.snapshot /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_tests/language/malformed_inheritance_test_08.dart
Took 0:00:00.196000

Command[vm]: out/DebugIA32Clang/fletch-vm /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_language_malformed_inheritance_test_08/fletch.snapshot
Did not run

Command[vm]: out/DebugIA32Clang/fletch-vm -Xunfold-program /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_language_malformed_inheritance_test_08/fletch.snapshot
Did not run

Short reproduction command (experimental):
    python tools/test.py -rfletchvm -cfletchc -t120 language/malformed_inheritance_test/08

FAILED: fletchc-fletchvm debug_ia32_clang language/malformed_inheritance_test/07
Expected: Pass 
Actual: Crash
Compile-time error expected.
CommandOutput[vm]:

stdout:
Using config file: /Users/ahe/Dart/fletch-repo/fletch/.fletch
out/DebugIA32Clang/generated_tests/language/malformed_inheritance_test_07.dart:14:17:
Warning: Additional type argument.
  <A> extends A<int> /// 07: compile-time error
                ^^^
out/DebugIA32Clang/generated_tests/language/malformed_inheritance_test_07.dart:14:15:
Error: Class 'C' can't extend the type 'A<int>' because it is malformed.
Try correcting the malformed type annotation or removing the 'extends' clause.
  <A> extends A<int> /// 07: compile-time error
              ^

Command[vm]: out/DebugIA32Clang/fletch -o /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_language_malformed_inheritance_test_07/fletch.snapshot /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_tests/language/malformed_inheritance_test_07.dart
Took 0:00:00.343000

Command[vm]: out/DebugIA32Clang/fletch-vm /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_language_malformed_inheritance_test_07/fletch.snapshot
Did not run

Command[vm]: out/DebugIA32Clang/fletch-vm -Xunfold-program /Users/ahe/Dart/fletch-repo/fletch/out/DebugIA32Clang/generated_compilations/fletchc/multitest_language_malformed_inheritance_test_07/fletch.snapshot
Did not run

Short reproduction command (experimental):
    python tools/test.py -rfletchvm -cfletchc -t120 language/malformed_inheritance_test/07

===
=== 2 tests failed
===

mkustermann, 2015-09-02:
Could this have been fixed? Maybe close this issue?

skabet, 2015-09-02:
3 months old with no update, closing.

closed 2015-09-02 by skabet