dart-archive/sdk #43: Running tests in certain mode causes 100% load with test timeouts

mkustermann, 2015-06-03:
Sometimes when running our tests in the following configuration I see a single fletch process which takes up 100% of my cpu and will continue to do so until it will be killed by the testing scripts. Running the test individually does not cause this behavior. One test on which this often happens is `tests/unsorted/messaging_benchmark_test.dart`.

I'm running the tests with "./tools/test.py -cfletchc -rfletchvm --asan --clang" on a z620.

This might indicate a bug in the scheduler. I've spent a number of hours looking into it, but could not identify the issue yet.

@skabet @madsager @kasperl Maybe someone with more knowledge about the scheduler code could look into this?

mkustermann, 2015-06-03:
Here are thread stacks of a process which behaves like that (taken with gdb):

[I can't attach it to the issue, so I need to paste it in here]

Thread 33 (Thread 0xf417cb40 (LWP 8067)):
#0  0xf7737430 in __kernel_vsyscall ()
#1  0xf76fcd4b in pthread_cond_wait@@GLIBC_2.3.2 () from /lib/i386-linux-gnu/libpthread.so.0
#2  0x082592c1 in fletch::PosixMonitor::Wait (this=<optimized out>) at ../../src/vm/platform_posix.cc:196
#3  0x083944ad in fletch::Scheduler::RunInThread (this=<optimized out>) at ../../src/vm/scheduler.cc:382
#4  0x0838e415 in fletch::Scheduler::RunThread (data=<optimized out>) at ../../src/vm/scheduler.cc:705
#5  0x08422398 in fletch::ThreadPool::RunThread (arg=<optimized out>) at ../../src/vm/thread_pool.cc:53
#6  0x080f2727 in ThreadStart () at /usr/local/google/work/chromium/src/third_party/llvm/projects/compiler-rt/lib/asan/asan_thread.cc:164
#7  0x080d748e in asan_thread_start () at /usr/local/google/work/chromium/src/third_party/llvm/projects/compiler-rt/lib/asan/asan_interceptors.cc:144
#8  0xf76f8f70 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#9  0xf74d3bee in clone () from /lib/i386-linux-gnu/libc.so.6

Thread 32 (Thread 0xf347cb40 (LWP 8068)):
#0  0x0821a5b3 in std::atomic<fletch::ThreadState*>::operator fletch::ThreadState* (this=0xf347ac20) at /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../include/c++/4.8/atomic:305
#1  0x0837d705 in fletch::ThreadState::next_idle_thread (this=<optimized out>) at ../../src/vm/process.h:48
#2  0x08393914 in fletch::Scheduler::PopIdleThread (this=<optimized out>) at ../../src/vm/scheduler.cc:360
#3  0x0839d792 in fletch::Scheduler::TryEnqueueOnIdleThread (this=<optimized out>, process=<optimized out>) at ../../src/vm/scheduler.cc:666
#4  0x08383a91 in fletch::Scheduler::EnqueueOnAnyThread (this=<optimized out>, process=<optimized out>, start_id=<optimized out>) at ../../src/vm/scheduler.cc:683
#5  0x08390890 in fletch::Scheduler::RescheduleProcess (this=<optimized out>, process=<optimized out>, state=<optimized out>, terminate=<optimized out>) at ../../src/vm/scheduler.cc:296
#6  0x0838b14d in fletch::Scheduler::InterpretProcess (this=<optimized out>, process=<optimized out>, thread_state=<optimized out>) at ../../src/vm/scheduler.cc:503
#7  0x083956f0 in fletch::Scheduler::RunInThread (this=<optimized out>) at ../../src/vm/scheduler.cc:413
#8  0x0838e415 in fletch::Scheduler::RunThread (data=<optimized out>) at ../../src/vm/scheduler.cc:705
#9  0x08422398 in fletch::ThreadPool::RunThread (arg=<optimized out>) at ../../src/vm/thread_pool.cc:53
#10 0x080f2727 in ThreadStart () at /usr/local/google/work/chromium/src/third_party/llvm/projects/compiler-rt/lib/asan/asan_thread.cc:164
#11 0x080d748e in asan_thread_start () at /usr/local/google/work/chromium/src/third_party/llvm/projects/compiler-rt/lib/asan/asan_interceptors.cc:144
#12 0xf76f8f70 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#13 0xf74d3bee in clone () from /lib/i386-linux-gnu/libc.so.6

Thread 31 (Thread 0xf2af8b40 (LWP 8069)):
#0  0xf7737430 in __kernel_vsyscall ()
#1  0xf76fcd4b in pthread_cond_wait@@GLIBC_2.3.2 () from /lib/i386-linux-gnu/libpthread.so.0
#2  0x082592c1 in fletch::PosixMonitor::Wait (this=<optimized out>) at ../../src/vm/platform_posix.cc:196
#3  0x083944ad in fletch::Scheduler::RunInThread (this=<optimized out>) at ../../src/vm/scheduler.cc:382
#4  0x0838e415 in fletch::Scheduler::RunThread (data=<optimized out>) at ../../src/vm/scheduler.cc:705
#5  0x08422398 in fletch::ThreadPool::RunThread (arg=<optimized out>) at ../../src/vm/thread_pool.cc:53
#6  0x080f2727 in ThreadStart () at /usr/local/google/work/chromium/src/third_party/llvm/projects/compiler-rt/lib/asan/asan_thread.cc:164
#7  0x080d748e in asan_thread_start () at /usr/local/google/work/chromium/src/third_party/llvm/projects/compiler-rt/lib/asan/asan_interceptors.cc:144
#8  0xf76f8f70 in start_thread () from /lib/i386-linux-gnu/libpthread.so.0
#9  0xf74d3bee in clone () from /lib/i386-linux-gnu/libc.so.6

Thread 30 (Thread 0xf2174b40 (LWP 8070)):
#0  0x0837d5a8 in fletch::ThreadState::next_idle_thread (this=0xf2172ea0) at ../../src/vm/process.h:48
#1  0x08393914 in fletch::Scheduler::PopIdleThread (this=<optimized out>) at ../../src/vm/scheduler.cc:360
#2  0x0839d792 in fletch::Scheduler::TryEnqueueOnIdleThread (this=<optimized out>, process=<optimized out>) at ../../src/vm/scheduler.cc:666
#3  0x08383a91 in fletch::Scheduler::EnqueueOnAnyThread (this=<optimized out>, process=<optimized out>, start_id=<optimized out>) at ../../src/vm/scheduler.cc:683
#4  0x08390890 in fletch::Scheduler::RescheduleProcess (this=<optimized out>, process=<optimized out>, state=<optimized out>, terminate=<optimized out>) at ../../src/vm/scheduler.cc:296
#5  0x0838b14d in fletch::Scheduler::InterpretProcess (this=<optimized out>, process=<optimized out>, thread_state=<optimized out>) at ../../src/vm/scheduler.cc:503
#6  0x083956f0 in fletch::Scheduler::RunInThread (this=<optimized out>) at ../../src/vm/scheduler.cc:413
#7  0x0838e415 in fletch::Scheduler::RunThread (data=<optimized out>) at ../../src/vm/scheduler.cc:705
#8  0x08422398 in fletch::ThreadPool::RunThread (arg=<optimized out>) at ../../src/vm/thread_pool.cc:53
#9  0x080f2727 in ThreadStart () at /usr/local/google/work/chromium/src/third_party/llvm/projects/compiler-rt/lib/asan/asan_thread.cc:164
[... cut ...]
#3  0x0838d146 in fletch::Scheduler::Run (this=<optimized out>) at ../../src/vm/scheduler.cc:232
#4  0x082c8c5f in fletch::Program::ProcessRun (this=<optimized out>, process=<optimized out>) at ../../src/vm/program.cc:1424
#5  0x0817f8e1 in fletch::RunSnapshot (bytes=...) at ../../src/vm/fletch_api_impl.cc:31
#6  0x0817eaca in FletchRunSnapshot (snapshot=<optimized out>, length=<optimized out>) at ../../src/vm/fletch_api_impl.cc:71
#7  0x08100d97 in fletch::Main (argc=<optimized out>, argv=<optimized out>) at ../../src/vm/main.

skabet, 2015-06-03:
Ah yes, I saw this a while ago, but I though it was fixed back then. It's the queue of idle threads that's misbehaving. I'll fix.

closed 2015-06-03 by skabet