dart-archive/sdk #44: Configurable package root

zerny, 2015-06-04:
Add a package root option to fletch/fletch-driver a package directory can be used from different locations.

Currently tests/service_tests/conformance/java/Makefile compiles snapshots with a different cwd than the fletch root and needs to be configures to use the package directory from the fletch root.

skabet, 2015-06-04:
Peter, I suppose we can do this as part of merging the driver/non-driver interface?

peter-ahe-google, 2015-06-04:
Yes, now seems like a good time to implement this.

skabet, 2015-07-06:
@zerny fixed the issue in c87486dbf9a1d065b30bce3dcce1b505a1ae2dbc.

peter-ahe-google, 2015-07-07:
Still not fixed for fletch (née fletch-driver).

peter-ahe-google, 2015-09-16:
@lukechurch I guess we need a UI for this. Any ideas? I would like something that works with incremental compilation, so I suggest:

    fletch set packages <file>

peter-ahe-google, 2015-09-18:
The solution is to have a settings file. It will be JSON and it contains a list of compile-time constants and the packages location.

I think this layout should suffice:

```json
{
  "packages": "relatative URI to .packages file",
  "constants": {
    "name": "value"
  }
}
```
When creating a session, one can specify a settings file:

```console
$ fletch create session my_session with settings.json
```

If no settings file is specified when creating a session, we use the file `.fletch-settings.json` if it exists.

peter-ahe-google, 2015-09-18:
Propose fix: [CL 1354253003](https://codereview.chromium.org/1354253003/)

peter-ahe-google, 2015-09-18:
Using this format:

```
{
  "package root": "A relative URI to the package root",
  "package root": "package",

  "options": "A list of strings that are passed to the compiler",
  "options": null,

  "contants": "A map of compile-time constants",
  "constants": null
}
```

skabet, 2015-09-19:
Why is this a settings file instead of a command line option?

peter-ahe-google, 2015-09-19:
The motivation for the settings file is that package root, compile-time constants (those from String.fromEnvironment, etc), and other compiler options need to be be part of the incremental compilation. When one of these values change, it can impact the output of the compiler.

I'm sure it is obvious to you that when the value of compile-time constant changes, we need to compute a new FletchSystem with those constants in mind. I think something similar happens when you change the package root, but in this case it is more like a bunch of files got added, removed, or modified. In either case, we need to compute a new FletchSystem.

So I think it is important that a user (a developer that uses Fletch) is aware that changing any of those settings changes the session/program/FletchSystem.

Another issue is that if these settings were controlled with options, when should you supply the options? The first time you compile? The second time? What happens if you forget an option? What happens if the persistent process crashes?

commit 23bd026d (2015-09-23): Implement session settings

Fixes #44

R=wibling@google.com

Review URL: https://codereview.chromium.org/1354253003 .
closed 2015-09-23 by peter-ahe-google