dart-archive/sdk #46: [test] Investigate slowness of corelib/int_parse_radix_test

mkustermann, 2015-06-08:
The 3 multitests started to timeout after most likely this CL: https://github.com/dart-lang/fletch/commit/ab74f83e76d480eaa59f0cf79f76b60eaef2d40b.

In order to reduce cycle times, this test is marked as Skip for all but one debug-x64 configuration.

mkustermann, 2015-06-11:
Small update:
@skabet made it already faster, but it is still too slow for debug-asan, so it is still marked as skip for some configurations in the status file.

skabet, 2015-06-18:
To some extend, this test is simply "broken". It performs 36^3*10 toString.

mit-mit, 2015-12-07:
@lrhn, we would love to fix this as part of FixIt (in the Dart SDK) and then pull it over to Fletch.

lrhn, 2015-12-08:
Stop being slow? :)
I'll look into doing fewer tests in the same run.

skabet, 2015-12-08:
Thank you, Lasse! :)

lrhn, 2015-12-10:
I have landed a CL that reduces the number of rounds, speeding up the test by ~30% on a desktop machine. If this is still too slow, I can have another look.

mit-mit, 2015-12-10:
@ricowind how do we get these changes pulled over to Fletch so we can see if the timeouts stop?

ricowind, 2015-12-10:
Basically:
https://github.com/dart-lang/fletch/wiki/Dart-Roll

@sigurdm @peter-ahe-google who have done rolls before, the last one was pretty painful afaik.

sigurdm, 2015-12-10:
Yes it was - maybe I should do the next one soon to amortize the pain...

sigurdm, 2015-12-10:
Alternatively we can make https://github.com/dart-lang/sdk/commit/fa025f33379119daa936a87a96c656e7e8e04623 as a patch on _temporary_fletch_patches

mit-mit, 2015-12-11:
Closing this for now, we will reopen if this turns out to be an issue after we roll.

closed 2015-12-11 by mit-mit