dart-archive/sdk #47: fletch-vm crash after SessionReset

peter-ahe-google, 2015-06-08:
This appears to be a crash after SessionReset.

Reproduce this by running:

`./tools/test.py -ax64 --clang --asan fletch_tests/incremental`

You may have to run many times.

```
FAILED: none-fletch_tests debug_x64_clang_asan fletch_tests/incremental/encoded/change_library_name
Expected: Pass 
Actual: Fail
CommandOutput[fletch_test]:

stdout:

Test 8 of 53 (0 skipped, 3 failed).
Full program #9:
==> main.dart <==
1: // Test that a change in library name is handled.
2: library test.main1;
3: 
4: main() {
5:   print('Hello, World!');
6: }

[verbose info] Using library root: file:///usr/local/google/home/ahe/fletch-repo/dart/sdk/
[verbose info] Using package root: org.trydart.packages:/
[verbose info] Allowed library categories: {Client, Shared, Internal}
[verbose info] Compiling org.trydart.compiler-test-case:/main.dart (build number could not be determined)
[verbose info] Scanning library org.trydart.compiler-test-case:/main.dart
[verbose info] Scanning library dart:core (file:///usr/local/google/home/ahe/fletch-repo/dart/sdk/lib/core/core.dart)
[verbose info] Scanning library dart:_fletch_system (file:///usr/local/google/home/ahe/fletch-repo/fletch/lib/system/system.dart)
[verbose info] Scanning library dart:_internal (file:///usr/local/google/home/ahe/fletch-repo/dart/sdk/lib/internal/internal.dart)
[verbose info] Scanning library dart:collection (file:///usr/local/google/home/ahe/fletch-repo/dart/sdk/lib/collection/collection.dart)
[verbose info] Scanning library dart:math (file:///usr/local/google/home/ahe/fletch-repo/dart/sdk/lib/math/math.dart)
[verbose info] Scanning library dart:convert (file:///usr/local/google/home/ahe/fletch-repo/dart/sdk/lib/convert/convert.dart)
[verbose info] Scanning library dart:async (file:///usr/local/google/home/ahe/fletch-repo/dart/sdk/lib/async/async.dart)
[verbose info] Scanning library dart:typed_data (file:///usr/local/google/home/ahe/fletch-repo/dart/sdk/lib/typed_data/typed_data.dart)
[verbose info] Scanning library dart:ffi (file:///usr/local/google/home/ahe/fletch-repo/fletch/lib/ffi/ffi.dart)
[verbose info] Resolving...
[verbose info] Resolved 461 elements.
[verbose info] Inferring types...
[verbose info] Compiling...
[verbose info] Compiled 283 methods.
[verbose info] Total compile-time 734msec; unaccounted 734msec
Running '/usr/local/google/home/ahe/fletch-repo/fletch/out/DebugX64ClangAsan/fletch-vm --port=39166'
Debugging
Instance of 'ProcessSpawnForMain'
PrepareForChanges()
PushFromMap(MapId.methods, 12)
PushBoolean(false)
ChangeMethodLiteral(0)
CommitChanges(1)
Debugging
Instance of 'ProcessSpawnForMain'
PushFromMap(MapId.methods, 12)
Instance of 'ProcessSetBreakpoint'
An error occurred, shutting down.
SocketException: Write failed, address = 127.0.0.1, port = 39166
#0      CommandReader.stream.<stream_async_gen_body> (package:fletchc/command_reader.dart)
#1      _asyncCatchHelper.<anonymous closure> (dart:core-patch/core_patch.dart:15)
#2      _rootRunBinary (dart:async/zone.dart:923)
#3      _CustomZone.runBinary (dart:async/zone.dart:819)
#4      _Future._propagateToListeners.handleError (dart:async/future_impl.dart:521)
#5      _Future._propagateToListeners (dart:async/future_impl.dart:580)
#6      _Future._completeError (dart:async/future_impl.dart:376)
#7      _StreamIteratorImpl._onError (dart:async/stream_impl.dart:1063)
#8      _rootRunBinary (dart:async/zone.dart:923)
#9      _CustomZone.runBinary (dart:async/zone.dart:819)
[... 145 lines cut ...]
#28     _rootRunUnary (dart:async/zone.dart:911)
#29     _CustomZone.runUnary (dart:async/zone.dart:811)
#30     _CustomZone.runUnaryGuarded (dart:async/zone.dart:717)
#31     _CustomZone.bindUnaryCallback.<anonymous closure> (dart:async/zone.dart:743)
#32     _NativeSocket.reportError (dart:io-patch/socket_patch.dart:976)
#33     _NativeSocket.write.<anonymous closure> (dart:io-patch/socket_patch.dart:648)
#34     _rootRun (dart:async/zone.dart:900)
#35     _CustomZone.run (dart:async/zone.dart:803)
#36     _CustomZone.runGuarded (dart:async/zone.dart:709)
#37     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#38     _rootRun (dart:async/zone.dart:904)
#39     _CustomZone.run (dart:async/zone.dart:803)
#40     _CustomZone.runGuarded (dart:async/zone.dart:709)
#41     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#42     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#43     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#44     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#45     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

Command[fletch_test]: FletchTestCommand(incremental/encoded/change_library_name)
Took 0:00:01.031375

Short reproduction command (experimental):
    python tools/test.py -ax64 --clang --asan -rfletch_tests -t60 fletch_tests/incremental/encoded/change_library_name
```

peter-ahe-google, 2015-06-08:
Notice that I've disabled this part of the test with https://codereview.chromium.org/1153793006/.

madsager, 2015-06-10:
I end up deleting the scheduler while another thread is still operating on it. A bit more synchronization is needed. Looking into that.

mkustermann, 2015-08-18:
@madsager  Can this issue be closed?

madsager, 2015-08-18:
Yes, session reset was removed. Thanks @mkustermann.

closed 2015-08-18 by madsager