dart-archive/sdk #49: ../../src/vm/object_memory.cc:105: error: expected: size >= HeapObject::kSize

lukechurch, 2015-06-10:
Considering the following rather long and very snarly code sample:

```
import 'dart:math' show Random;

class Mutable {
  int x;
}

entry(fn) => fn();

bool isArgumentError(o) => o is ArgumentError;

main() {

  Random r = new Random();

  List dataSet = [];
  for (int i = 0; i < 1000; i++) {
    var pt = new ImmutablePoint(ImmutableLinkedList.fromList(
      [r.nextInt(2),r.nextInt(2), r.nextInt(2)]));
      dataSet.add(pt);
  }

  for (int i = 0; i < 1000; i++) {
    var pt = new ImmutablePoint(ImmutableLinkedList.fromList(
      [r.nextInt(2) + 5,r.nextInt(2) +5 , r.nextInt(2) + 5]));
      dataSet.add(pt);
  }

  var centroidsList = ImmutableLinkedList.fromList([
    new ImmutablePoint(ImmutableLinkedList.fromList([-10,0,0])),
    new ImmutablePoint(ImmutableLinkedList.fromList([0,15,0])),
    ]);

  allocate(ImmutablePoint pt) {
    int bestDistance = 2147483647;
    ImmutablePoint allocation = null;

    for (var centroid in centroidsList.toMutable()) {
      int distance = centroid.subtract(pt).magnitudeSq;

      if (distance < bestDistance) {
        allocation = centroid;
        bestDistance = distance;
      }
    }
    return new ImmutableAllocation(pt, allocation);
  }

print ('about to run');
  print( parallel.map(allocate)(dataSet));
}

const Parallel parallel = const Parallel();

class Parallel implements Function {
  const Parallel();
  Iterable call(Iterable values) => values;
  Parallel map(fn(value)) => new _ParallelMap(this, fn);
}

class _ParallelMap extends Parallel {
  final Parallel _link;
  final Function _fn;
  const _ParallelMap(this._link, this._fn);

  Iterable call(Iterable values) {
    final Function fn = _fn;
    List fns = _link(values).map((final e) => () => fn(e)).toList();
    return Process.divide(_entry, fns);
  }

  static _entry(fn) => fn();
}

class ImmutablePoint {
  final ImmutableLinkedList coords;
  const ImmutablePoint(this.coords);

  ImmutableVector subtract(ImmutablePoint pt) {

    List localCoords = coords.toMutable();
    List ptCoords = pt.coords.toMutable();

    List ret = [];
    for (int i = 0; i < localCoords.length; i++){
      ret.add(localCoords[i] - ptCoords[i]);
    }
    return new ImmutableVector(ImmutableLinkedList.fromList(ret));
  }

  toString() {
    return coords.toMutable().toString();
  }
}

class ImmutableVector {
  final ImmutableLinkedList args;
  const ImmutableVector(this.args);

  get magnitudeSq {
    ImmutableLinkedList pointer = args;
    int acc = 0;

    while (pointer != null) {
      var data = pointer.data;
      acc += data * data;
      pointer = pointer.next;
    }
    return acc;
  }

  String toString() {
    return args.toString();
  }
}

class ImmutableLinkedList {
  final ImmutableLinkedList next;
  final int data;
  const ImmutableLinkedList(this.data, this.next);

  toString() {
    StringBuffer sb = new StringBuffer();
    ImmutableLinkedList pointer = this;
    while (pointer.next != null) {
      sb.write("${pointer.data}, ");
      pointer = pointer.next;
    }
    sb.write("${pointer.data}");
    return sb.toString();
  }

  static ImmutableLinkedList fromList(List data) {
    var reversed = data.reversed.toList();
    ImmutableLinkedList node = new ImmutableLinkedList(reversed.first, null);
    for (int i = 1; i < data.length; i++) {
      node = new ImmutableLinkedList(reversed[i], node);
    }
    return node;
  }

  List toMutable() {
    List builder = [];
    ImmutableLinkedList pointer = this;
    while (pointer.next != null) {
      builder.add(pointer.data);
      pointer = pointer.next;
    }
    builder.add(pointer.data);
    return builder;
  }
}

class ImmutableAllocation {
  final ImmutablePoint pt;
  final ImmutablePoint centroid;

  const ImmutableAllocation(this.pt, this.centroid);

  toString() => "$pt -> $centroid";
}```

If you remove the last line overriding toString, execute completes.

With the addition of this line, the result becomes:

../../src/vm/object_memory.cc:105: error: expected: size >= HeapObject::kSize
Non-zero exit code from '/Users/lukechurch/GitRepos/fletch-repo/fletch/out/DebugX64Clang/fletch-vm' (-6).
Abort trap: 6

madsager, 2015-06-10:
Luke, I would like to have a look at this but I can't reproduce with current tip of tree. Can you still reproduce this? If so, can you provide more detail on how you are running it?

lukechurch, 2015-06-10:
@madsager Interesting, I can reproduce this on the branch associated with this CL: https://codereview.chromium.org/1177583002 but not on tip of tree. Which presumably means that its something to do with the way I'm generating the numbers that will hopefully go away when I address Anders' suggestion of a Dart only implementation. 

However, I don't see why that change would cause this issue. I'm fine with closing for now and reopening, otherwise repo steps are to pull that branch, save the contents to km.dart and run on OS X the equivalent of:

~/GitRepos/fletch-repo/fletch/out/DebugX64Clang/fletch compile-and-run km.dart

madsager, 2015-06-10:
Thanks Luke, that gives me what I need. I'm pretty sure this has got nothing to do with your patch. I think it is a GC issue in connection with the Process.divide primitive. Your patch leads to enough allocation that we hit the issue. We are getting a pointer to an ImmutablePoint that is in a heap chunk that we have already deleted.

madsager, 2015-06-10:
Thanks for reporting this issue Luke, I now understand what is going on. The merging of heaps involved in Process.divide doesn't work at this point. The merging is controlled via Dart code in the parent process. However, that means that the parent process wakes up and executes Dart code while it is merging in the heaps of the children. That can lead to a GC of the parent process before all the child heaps have been merged in. The child heaps that have not been merged in will have stale pointers to immutable data in the parent if that happens. We need to not allow Dart code to run in the parent before we have merged in all the child heaps. Working on that. :)

closed 2015-06-11 by madsager

lukechurch, 2015-06-11:
Great. Thanks @madsager