dart-archive/sdk #61: Document how to roll a new version of Dart

peter-ahe-google, 2015-06-29:
Ultra short version is:

Checkout this branch:

https://github.com/dart-lang/sdk/tree/_temporary_fletch_patches

Identify the commit on the master branch you want, often just:

```git rev-parse origin/master```

Merge this commit into _temporary_fletch_patches using --no-edit:

```git merge -X theirs --no-commit COMMIT_ID```

Make the diff is empty:

```git checkout COMMIT_ID -- .```

Verify that the diff is empty:

```git diff COMMIT_ID```

Commit the merge:

```git commit --no-edit```

Identify the patches needed:

```git log --first-parent --oneline origin/_temporary_fletch_patches..HEAD```

Cherry-pick the changes:

```git cherry-pick <commits>```

Push the changes:

```git push origin _temporary_fletch_patches```

kasperl, 2015-07-07:
Once you push the changes, we have 3 dart-sdk-fletch-patched-* bots building the patched Dart VM binaries for Mac OS, Linux, and Linux ARM and uploading them to Google Cloud Storage.

Here are examples of the produced binaries:

https://storage.googleapis.com/fletch-archive/91d809c99f6d80b58f5e85782e4bc18f7f9efb53/dart-vm-x64-macos
https://storage.googleapis.com/fletch-archive/91d809c99f6d80b58f5e85782e4bc18f7f9efb53/dart-vm-x64-linux
https://storage.googleapis.com/fletch-archive/91d809c99f6d80b58f5e85782e4bc18f7f9efb53/dart-vm-arm-linux

We may still need to tweak the build settings to make sure the built binaries work in all the settings we need them to work in.

skabet, 2015-07-22:
Can confirm the output is not glibc compatible for ARM.

mkustermann, 2015-07-22:
We need to build the dart binary on an ARM machine I think or on a older linux machine if we want to cross compile.

peter-ahe-google, 2015-08-14:
Fixed in https://github.com/dart-lang/fletch/wiki/Dart-Roll/78edff0974f3fc4f733a0f2f1927d11f5bba6ebe

closed 2015-08-14 by peter-ahe-google