dart-archive/sdk #62: Need a better error message for mirrors

lukechurch, 2015-06-30:
The current error message is the standard one from Dart2JS, which refers to a compiler the user didn't run.

We should replace it with something like the below

```
*********************************************************************
* dart:mirrors is not currently supported by the fletch runtime.
* We want to try something different in this space using a
* persistent compiler process. Please see XXXX for more
* details.
* 
* Your app imports dart:mirrors via:
*   .....dart => dart:mirrors
*********************************************************************
```

lukechurch, 2015-06-30:
And this is an even worse error message - which I think is saying the same thing.

```
Error: External function is not supported
  external Type get runtimeType;
  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Unhandled exception:
Uncaught Error: Unsupported constant: TypeConstant(String)
Stack Trace:
#0      FletchSystemBuilder.computeSystem.<anonymous closure> (package:fletchc/src/fletch_system_builder.dart:259:9)
#1      _CompactLinkedHashMap.forEach (dart:collection-patch/compact_hash.dart:289)
#2      FletchSystemBuilder.computeSystem (package:fletchc/src/fletch_system_builder.dart:199:19)
#3      FletchBackend.computeDelta (package:fletchc/src/fletch_backend.dart:1074:41)
#4      FletchCompiler.run.<run_async_body>.<anonymous closure> (package:fletchc/compiler.dart:227:54)
#5      _RootZone.runUnary (dart:async/zone.dart:1155)
#6      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#7      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#8      _Future._completeWithValue (dart:async/future_impl.dart:368)
#9      _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#10     _asyncRunCallbackLoop (dart:async/schedule_microtask.dart:41)
#11     _asyncRunCallback (dart:async/schedule_microtask.dart:48)
#12     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#13     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:143)

#0      _rootHandleUncaughtError.<anonymous closure> (dart:async/zone.dart:886)
#1      _asyncRunCallbackLoop (dart:async/schedule_microtask.dart:41)
#2      _asyncRunCallback (dart:async/schedule_microtask.dart:48)
#3      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#4      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:143)
```

peter-ahe-google, 2015-09-16:
Proposed fix: [CL 1345203002](https://codereview.chromium.org/1345203002/)

When compiling this file:

```
import 'dart:mirrors';

main() {
  print(new Object().runtimeType);
  print("Fisk");
}
```

The output is:

```
mirrors.dart:
Error: Fletch has no support for 'dart:mirrors'. See https://goo.gl/Kwrd0O

mirrors.dart:4:9:
Error: 'runtimeType' isn't supported in Fletch. See https://goo.gl/ELH6Zc
  print(new Object().runtimeType);
        ^^^^^^^^^^^^^^^^^^^^^^^^
```

commit a7d4af04 (2015-09-18): Improve various error messages

Fixes #62

R=johnniwinther@google.com, lukechurch@google.com

Review URL: https://codereview.chromium.org/1345203002 .
closed 2015-09-18 by peter-ahe-google