dart-archive/sdk #65: language/non_parameterized_factory2_test might be flaky

peter-ahe-google, 2015-06-30:
When testing commit 2ab69c34a8211b242d457cc85ddc349be787abca, fletch-mac-debug-x86 found the problem below.

It is highly unlikely that said commit is caused this problem, and the test succeeded on the next build (which was due to commit f53b7b256fdf5dce5a0ed80a1c9a672b6a0a4fe6, which conversely is unlikely to have fixed a problem in this test).

Exit code -10 means that the VM died due to SIGBUS.

```
@@@BUILD_STEP test_DebugIA32Clang failures@@@

FAILED: none-fletchc debug_ia32_clang language/non_parameterized_factory2_test
Expected: Pass 
Actual: Crash
CommandOutput[vm]:

stdout:
tests/language/non_parameterized_factory2_test.dart:7:38:
Warning: '(dynamic) -> Factory<T>' is not assignable to '(T) -> Interface<T>'.
  factory Interface.withArg(T value) = Factory<T>.withArg;
                                     ^^^^^^^^^^^^^^^^^^^^^
tests/language/non_parameterized_factory2_test.dart:6:23:
Warning: '() -> Factory<T>' is not assignable to '() -> Interface<T>'.
  factory Interface() = Factory<T>;
                      ^^^^^^^^^^^^^
Non-zero exit code from '/Volumes/data/b/build/slave/fletch-mac-debug-x86/build/fletch/out/DebugIA32Clang/fletch-vm' (-10).

Command[vm]: out/DebugIA32Clang/fletch compile-and-run /Volumes/data/b/build/slave/fletch-mac-debug-x86/build/fletch/tests/language/non_parameterized_factory2_test.dart
Took 0:00:01.581000

Short reproduction command (experimental):
    python tools/test.py --kill-persistent-process 0 --run-gclient-hooks 0 --build-before-testing 0 -rfletchc -t120 language/non_parameterized_factory2_test
```

kasperl, 2015-07-09:
Looks like we have another instance of this happening in a different test:

https://uberchromegw.corp.google.com/i/client.dart.internal/builders/fletch-mac-release-x86/builds/141/steps/Test%20ReleaseIA32Clang/logs/stdio

@peter-ahe-google Should we update the issue "title" to reflect that this might be a more general issue?

```
FAILED: none-fletchc release_ia32_clang language/factory_arrow_test
Expected: Pass 
Actual: Crash
CommandOutput[vm]:

stdout:
Non-zero exit code from '/b/build/slave/fletch-mac-release-x86/build/fletch/out/ReleaseIA32Clang/fletch-vm' (-10).

Command[vm]: out/ReleaseIA32Clang/fletch compile-and-run /b/build/slave/fletch-mac-release-x86/build/fletch/tests/language/factory_arrow_test.dart
Took 0:00:01.659000

Short reproduction command (experimental):
    python tools/test.py -mrelease --kill-persistent-process 0 --run-gclient-hooks 0 --build-before-testing 0 -rfletchc -t60 language/factory_arrow_test

Done none-fletchc release_ia32_clang language/factory_arrow_test: fail
```

peter-ahe-google, 2015-07-09:
It could be unrelated problems. I suggest we track this in separate issues until we have more information.

mkustermann, 2015-09-02:
Since https://github.com/dart-lang/fletch/commit/5d5998aec843da31cf03fb84069ec65967c59621 we have a proper shutdown mechanism in the vm (in case of normal shutdowns). The exit code `-10 = SIGBUS` indicates that this was the issue.

closed 2015-09-02 by mkustermann