dart-archive/sdk #69: Correct shutdown handling of fletch-vm in all circumstances

mkustermann, 2015-07-09:
There are 3 cases where shutdown is not properly handled

  * Waiting for session message handling thread to finish
  * Uncaught errors just `exit()`
  * Compile time errors just `exit()`

If the main thread returns or an arbitrary thread calls `exit()`, the program will start running `atexit()` handlers ... This can cause issues in other threads. The other threads might then want to `FATAL` something, which invokes `printf` which doesn't work (can e.g. call a SIGBUS error) after `atexit()` handlers were invoked.

If the fletch-vm is embedded into a larger application, calling `exit()` will bring down more than just the fletch-vm. Rather than doing that, the fletch-vm should allow the embedder to detect and handle the error.

Here are more details to these 3 cases:

Waiting for session to finish
=== 

The code in `src/vm/main.cc` looks like:

```c++
 18 static bool RunSession(Connection* connection) {
 19   Session session(connection);
 20   session.Initialize();
 21   session.StartMessageProcessingThread();
 22   return session.ProcessRun();
 23 } 
```

Once `ProcessRun()` is done running, the `Session` object goes out of scope and it's destructor is called which deletes the connection object:

```c++
  43 Session::~Session() {
  44   delete connection_;
  45   delete program_->scheduler();
  46   delete program_;
  47   delete main_thread_monitor_;
  48   for (int i = 0; i < maps_.length(); ++i) delete maps_[i];
  49   maps_.Delete();
  50 }
```

At this point, the `connection_` object is dead. We are not explicitly waiting for the session thread to be done e.g. via `pthread_join`. 

Uncaught errors just `exit()`
=== 
The interpreter will try to find an error handler, and if it does not succeed, it might just `exit(1)` - leaving all other threads in a possible inconsistent state, `src/vm/interpreter.cc`:

```c++
1133 uint8* HandleThrow(Process* process, Object* exception, int* stack_delta) {
...
1145       // Send stack trace information to attached session if debugging.  
1146       Session* session = process->program()->session();      
1147       if (session != NULL && session->is_debugging()) {      
1148         session->UncaughtException();
1149         return NULL;
1150       }
1151       exit(1);
1152     }
```

Compile time errors just `exit()`
===
The fletchc compiler will emit a native call to `halt` which just exits the program, leaving all other threads in a possible inconsistent state, `src/vm/natives.cc`:

```c++
  45 NATIVE(Halt) {
  46   word value = Smi::cast(arguments[0])->value();
  47   if (value == 0) return Failure::should_preempt();
  48   exit(value);
  49 }

```

This **might** have caused flaky `SIGBUS` errors on our buildbots. It's unclear whether this is the case. But it seems advisable to fix these issues in any case.

mkustermann, 2015-07-09:
Might have caused SIGBUS issues described in #65 .

mkustermann, 2015-07-21:
@madsager took care of the two `exit()` issues. We are still not using `pthread_join()` to wait for termination of other threads. But that might not be a big deal ATM.

mkustermann, 2015-08-06:
Since 5d5998aec843 we're using `pthread_join()` now. 

We're still calling just `exit()` in the scheduler for uncaught exceptions / compile-time errors. In case the fletch-vm is e.g. embedded in a larger application it seems not a good solution. It should be possible to notify the embedder that an non recoverable error occurred in a process.

mit-mit, 2015-12-07:
@skabet thinks he fixed this, @mkustermann we are going to close this, please reopen if you are still seeing it.

closed 2015-12-07 by mit-mit