dart-archive/sdk #75: Driver should support forwarding -X flags to fletch-vm

mkustermann, 2015-07-17:
We have a number of tests which use "// FletchOptions=-X..." which the testing script forwards to the running command. The driver does not understand these flags and crashes.

peter-ahe-google, 2015-09-07:
The testing script can easily forward these options to the fletch VM. Generally, the driver is moving away from starting the VM directly, so I need a bit of clarification: Is this bug about when we run our tests, or is this about `compile-and-run`?

mkustermann, 2015-09-07:
See this status file `tests/unsorted/unsorted.status`:
```
[ $compiler == none && $runtime == fletchc ]
immutable_and_mutable_heap_validate_heaps_test: CompileTimeError # Issue(75): Driver doesn't support -X flags.
messaging_drop_messages_expose_gc_test: CompileTimeError # Issue(75): Driver doesn't support -X flags.
messaging_port_lifetime_expose_gc_test: CompileTimeError # Issue(75): Driver doesn't support -X flags.
messaging_process_die_early_gc_on_delete_test: CompileTimeError # Issue(75): Driver doesn't support -X flags.
messaging_program_gc_gc_on_delete_test: CompileTimeError # Issue(75): Driver doesn't support -X flags.
```

Trying to run it gives
```
FAILED: none-fletchc release_ia32 unsorted/messaging_port_lifetime_expose_gc_test
Expected: Pass 
Actual: CompileTimeError
CommandOutput[fletch_session]:

stderr:
Compiler options not implemented: -Xexpose-gc

Command[fletch_session]: /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tools/testing/bin/linux/dart -c -Dtest.fletch_session_command.is_incremental_enabled=false tools/testing/dart/fletch_session_command.dart out/ReleaseIA32/fletch /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/unsorted/messaging_port_lifetime_expose_gc_test.dart -Xexpose-gc /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/unsorted/messaging_port_lifetime_expose_gc_test.dart
Took 0:00:00.000000

Short reproduction command (experimental):
    python tools/test.py -mrelease -rfletchc -t60 unsorted/messaging_port_lifetime_expose_gc_test

```

peter-ahe-google, 2015-09-07:
In that case we need to change `tools/testing/dart/fletch_session_command.dart`.

mit-mit, 2015-12-07:
@peter-ahe-google is this a FixIt candidate?

peter-ahe-google, 2015-12-07:
I don't think our users care about this. What does @mkustermann and @skabet think?

mkustermann, 2015-12-10:
The bugreport is not about what our users care about, but rather about our test infrastructure.

The purpose of those tests is to pass `-X` flags to the fletch-vm in order test certain VM behavior. I am of the opinion that we **need** to be able to do this!

The question is:
  * Should we limit ourselves to only run `-X` tests via `--compiler=fletchc --runtime=fletchvm` (i.e. snapshot runs)?

   => If so, we can just Skip these tests in the status file.

  * Should we support running these tests also with the normal driver?

   => If so, we should make sure the `-X` flags get passed to the `fletch-vm` instead of the `fletch run`.

peter-ahe-google, 2016-01-04:
Since this is for our own testing, then we need to modify `spawnVm()` to accept the `-X`-arguments in `tools/testing/dart/fletch_session_command.dart`. The driver doesn't start fletch-vm during automated testing, the test framework (`fletch_session_command.dart`) does.

In addition, the method `FletchSessionCommand.run` would need to accept `-X`-arguments. Right now it simulates a compiler error if there are more than on command line argument to a test.