dart-archive/sdk #79: language/closure_in_initializer2_test causes SEGV

peter-ahe-google, 2015-07-21:
This was observed in when building 12cb06be9bd4992caec3bd25f04e9762428d2278 on fletch-mac-debug-x86 (its build 100).

Since then, I've not seen the problem repeat.

```
FAILED: none-fletchc debug_x64_clang language/closure_in_initializer2_test
Expected: Pass 
Actual: Crash
CommandOutput[vm]:

stdout:
Non-zero exit code from '/Volumes/data/b/build/slave/fletch-mac-debug-x86/build/fletch/out/DebugX64Clang/fletch-vm' (-11).

Command[vm]: out/DebugX64Clang/fletch compile-and-run /Volumes/data/b/build/slave/fletch-mac-debug-x86/build/fletch/tests/language/closure_in_initializer2_test.dart
Took 0:00:01.657000

Short reproduction command (experimental):
    python tools/test.py -ax64 --kill-persistent-process 0 --run-gclient-hooks 0 --build-before-testing 0 -rfletchc -t120 language/closure_in_initializer2_test
```

mkustermann, 2015-09-02:
@skabet Did you look into it?

If not: It doesn't seem like this has happened since then, so maybe we should close this github issue?

skabet, 2015-09-02:
I did some changes, so maybe :) I'll close for now.

closed 2015-09-02 by skabet