dart-archive/sdk #80: Break on throw - don't wait until all coroutine stacks have been unwound

mkustermann, 2015-07-22:
Currently our debugging protocol allows us to spawn a new process, set breakpoints and run the process -- which can result in either one of the following events:

1. a breakpoint was reached
2. a compile-time error was hit
3. an uncaught exception occurs
4. process terminates normally

In cases *1. breakpoint was reached* and *2. compile-time error*, the interpreter returns immediately to the scheduler which in return transfers control to the session. Then the debugger can inspect the stack and possibly change code and/or restart a frame.

In cases *4. process terminates normally* there is no stack left, and no reason to continue running the process.

In case *3. an uncaught exception occurs*, the coroutine call chain got unwound. (The last coroutine stack might still be left.) If finally blocks were found they are probably executed.

=> This might be rather unfortunate for a live coding experience: At the point in time when the debugger signals an uncaught exception, most/all frames might have been already unwound, making it impossible to do a program change and restart a frame. We might want to consider having a debugging mode where **all throw expressions** will transfer to control to the debugger - without unwinding all coroutine stacks first - to allow developers to "selectively unwind frame by frame", "live edit" and "restart a frame".

WDYT?

cc: @madsager, @peter-ahe-google

peter-ahe-google, 2015-07-23:
Agreed, we need to be able to pause on all throws (before unwinding the
stack).

We probably also need more, for example, a way to not pause on exceptions
that are known to be handled (although this is rather tricky in relation to
async).

Possibly, creating a heap snapshot on throw will allow us to step back in
time to before the stack was unwound.

madsager, 2015-07-23:
Uncaught exceptions do work at this point and the stack is *not* unwound for uncaught exceptions. I have debugger tests that breaks on an uncaught exception and restarts. Here is an example:

```dart
bar() {
  print('bar');
  throw 42;
}

foo() {
  print('foo');
  bar();
}

main() {
  foo();
}
```

Debugger session showing that we do *not* unwind on uncaught exceptions.

```
$ out/DebugIA32Clang/dart -c -p package package:fletchc/fletchc.dart -d test.dart
Starting session. Type 'help' for a list of commands.

> r
foo
bar
Uncaught exception:
42

Stack trace:
> 0: bar     	@42
  1: foo     	@bar()
  2: main    	@foo()
> restart
bar
Uncaught exception:
42

Stack trace:
> 0: bar     	@42
  1: foo     	@bar()
  2: main    	@foo()
> q
```

It definitely makes sense to have a break on all exceptions setting. It is usually not very useful for programs where exceptions are used for control flow. In that case it becomes too tedious to find the throw that you are interested in. But for some users this could probably be very useful.

peter-ahe-google, 2015-07-23:
Unfortunately, if I  modify the program to this:

```
bar() {
  print('bar');
  throw 42;
}

foo() {
  print('foo');
  bar();
}

main() {
  try {
    foo();
  } catch (e) {
    rethrow;
  }
}
```

The stack trace becomes:

```
Stack trace:
> 0: main    	@foo()
```

There's a disconnect between what a VM engineer considers an uncaught exception, and what a developer considers an uncaught exception.

madsager, 2015-07-23:
Acknowledged. It would make sense to have a 'break on exception' mode where you break when you encounter a throw without doing anything.

The other interesting case that Martin is pointing out and that I missed on a first reading is the way unwinding works for coroutines. Currently, we destroy the stack for the coroutines and therefore we will end up with a nonsensical stack trace for an uncaught exception in a coroutine:

```dart
import 'dart:fletch';

bar() {
  print('bar');
  throw 42;
}

foo(i) {
  print('foo');
  bar();
}

main() {
  new Coroutine(foo)(0);
}
```

For the above code the implementation will throw away all the linked coroutine stacks and you will end back at main which is not what the user would expect:

```
$ out/DebugIA32Clang/dart -c -p package package:fletchc/fletchc.dart -d test.dart
Starting session. Type 'help' for a list of commands.

> r
foo
bar
Uncaught exception:
42

Stack trace:
> 0: main          	@new Coroutine(foo)(0)  // WAT?!?
> q
```

I my mind the current behavior makes sense for Fibers (where we get the break on *uncaught* exception behavior that users should be used to). In addition to that we should probably add a break on *any* exception option. Finally, we should make sure that all of the above works properly not only for Fibers but for coroutines as well.

peter-ahe-google, 2015-07-23:
I'm strongly of the opinion that the current behavior for uncaught exceptions doesn't make sense, but I agree users may have been trained to accept it. However, that doesn't make it an attractive feature.

peter-ahe-google, 2015-07-23:
@lukechurch

mkustermann, 2015-10-02:
The original issue - about accidentally unwinding coroutine stacks - has been fixed.

We can have another discussion at some point about: finally, rethrow and snapshots.

closed 2015-10-02 by mkustermann