dart-archive/sdk #84: ASAN heap-use-after-free  in feature_test.dart

peter-ahe-google, 2015-07-29:
Error message from ASAN:

```
=================================================================
==5163==ERROR: AddressSanitizer: heap-use-after-free on address 0xf5401bb0 at pc 0x081e9b52 bp 0xf51feef8 sp 0xf51feef0
READ of size 4 at 0xf5401bb0 thread T1
```

Full test log output:

```
FAILED: none-fletch_tests debug_ia32_clang_asan fletch_tests/incremental/encoded/two_updates
Expected: Pass 
Actual: Fail
CommandOutput[fletch_test]:

stdout:

Test 2 of 53 (0 skipped, 0 failed).
Full program #3:
==> main.dart <==
1: main() { print('Hello darkness, my old friend'); }
[verbose info] Using library root: file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/
[verbose info] Using package root: org.trydart.packages:/
[verbose info] Allowed library categories: {Client, Shared, Internal}
[verbose info] Compiling org.trydart.compiler-test-case:/main.dart (build number could not be determined)
[verbose info] Scanning library org.trydart.compiler-test-case:/main.dart
[verbose info] Scanning library dart:core (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/core/core.dart)
[verbose info] Scanning library dart:_fletch_system (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/lib/system/system.dart)
[verbose info] Scanning library dart:_internal (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/internal/internal.dart)
[verbose info] Scanning library dart:collection (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/collection/collection.dart)
[verbose info] Scanning library dart:math (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/math/math.dart)
[verbose info] Scanning library dart:fletch (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/lib/fletch/fletch.dart)
[verbose info] Scanning library dart:convert (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/convert/convert.dart)
[verbose info] Scanning library dart:async (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/async/async.dart)
[verbose info] Scanning library dart:fletch.io (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/lib/io/io.dart)
[verbose info] Scanning library dart:fletch.ffi (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/lib/ffi/ffi.dart)
[verbose info] Scanning library dart:typed_data (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/dart/sdk/lib/typed_data/typed_data.dart)
[verbose info] Scanning library dart:system (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/lib/io/system.dart)
[verbose info] Resolving...
[verbose info] Resolved 457 elements.
[verbose info] Inferring types...
[verbose info] Compiling...
[verbose info] Compiled 283 methods.
[verbose info] Total compile-time 1137msec; unaccounted 1137msec
Running '/mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/out/DebugIA32ClangAsan/fletch-vm -Xvalidate-heaps --port=42352'
CommandCode.Debugging()
CommandCode.ProcessSpawnForMain()
breakpoint set: 0: main @0
Stack trace:
> 0: main    	@'Hello darkness, my old friend'
  1: callMain	@invokeMain(arguments)
  2: entry   	@callMain([] )
fletch_vm_stdout: Hello darkness, my old friend
Stack trace:
> 0: callMain	@=>invokeMain(arguments);
  1: entry   	@callMain([] )
Got expected output: Hello darkness, my old friend
Update:
==> main.dart <==
1: main() { print('I\'ve come to talk with you again'); }
Reusing origin library(dart:core) (assumed read-only).
Reusing library(dart:_fletch_system) (assumed read-only).
Reusing origin library(dart:_internal) (assumed read-only).
Reusing origin library(dart:collection) (assumed read-only).
Reusing origin library(dart:math) (assumed read-only).
Reusing library(dart:fletch) (assumed read-only).
Reusing origin library(dart:convert) (assumed read-only).
Reusing origin library(dart:async) (assumed read-only).
Reusing library(dart:fletch.io) (assumed read-only).
[... 90 lines cut ...]
=================================================================
==5163==ERROR: AddressSanitizer: heap-use-after-free on address 0xf5401bb0 at pc 0x081e9b52 bp 0xf51feef8 sp 0xf51feef0
READ of size 4 at 0xf5401bb0 thread T1

#0      TestSession.waitForCompletion.<waitForCompletion_async_body> (file:///mnt/data/b/build/slave/fletch-linux-asan-x86/build/fletch/tests/fletchc/incremental/feature_test.dart:2481:7)
#1      _rootRunUnary (dart:async/zone.dart:911)
#2      _CustomZone.runUnary (dart:async/zone.dart:811)
#3      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#4      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#5      _Future._addListener.<anonymous closure> (dart:async/future_impl.dart:274)
#6      _rootRun (dart:async/zone.dart:904)
#7      _CustomZone.run (dart:async/zone.dart:803)
#8      _CustomZone.runGuarded (dart:async/zone.dart:709)
#9      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#10     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#11     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#12     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#13     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

Command[fletch_test]: FletchTestCommand(incremental/encoded/two_updates)
Took 0:00:01.544571

Short reproduction command (experimental):
    python tools/test.py --kill-persistent-process 0 --run-gclient-hooks 0 --build-before-testing 0 --asan --clang -rfletch_tests -t60 fletch_tests/incremental/encoded/two_updates
```

peter-ahe-google, 2015-07-29:
This was fletch-linux-asan-x86 Build #177 for commit 8ae3a6e37b04495a1e52e4ead9a1c8906e0dcf9e

peter-ahe-google, 2015-07-29:
This appears flaky to be flaky.

ricowind, 2015-07-30:
I have tried reproducing this locally, no luck

closed 2015-12-07 by mit-mit