dart-archive/sdk #92: Errors in SharedTask objects crash hosting session

smanilov, 2015-08-12:
If a `task` that is performed in a worker thread in the persistent process throws an unhandled error, the hosting session crashes. If the session is not `x-end`ed, then trying to run another task in it will freeze the client process.

Example code:

    class BadTask extends SharedTask {
        Future<int> call(CommandSender s, StreamIterator<Command> i) {
            return new Future<int>(() => throw new Error());
        }
    }

    // executing this will crash the current session
    context.performTaskInWorker(new BadTask());

peter-ahe-google, 2015-08-12:
This is related to this TODO: https://github.com/dart-lang/fletch/blob/1cb07ac8050ccfad3148f81fb061d3bac7cdf19e/pkg/fletchc/lib/src/driver/driver_main.dart#L504

peter-ahe-google, 2015-08-12:
FYI: for exiting the service compiler due to an error, without crashing the session, consider using throwFatalError (https://github.com/dart-lang/fletch/blob/1cb07ac8050ccfad3148f81fb061d3bac7cdf19e/pkg/fletchc/lib/src/diagnostic.dart#L150)

smanilov, 2015-08-12:
Okay, will have it in mind, thanks!

peter-ahe-google, 2015-09-23:
Removing accepted for now to signal that this isn't as high priority as other accepted bugs in Tools.

mit-mit, 2015-12-07:
@peter-ahe-google is this something we can won't fix? Doesn't sound that important?

peter-ahe-google, 2015-12-07:
It isn't super important, but it is a problem. I'm not a big fan of closing bug reports when the underlying issue isn't fixed. Is there a label we can use for this?

closed 2016-05-30 by mit-mit