dart-archive/sdk #93: ASAN build might fail on mac

zerny, 2015-08-12:
The build dependencies of natives_to_json should depend on copy_asan.

```
FAILED: cd ../../src/shared; export BUILT_PRODUCTS_DIR=/Users/zerny/src/fletch/fletch/out/DebugIA32ClangAsan; export CONFIGURATION=DebugIA32ClangAsan; export PRODUCT_NAME=natives_json; export SDKROOT=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk; export SRCROOT=/Users/zerny/src/fletch/fletch/out/DebugIA32ClangAsan/../../src/shared; export SOURCE_ROOT="${SRCROOT}"; export TARGET_BUILD_DIR=/Users/zerny/src/fletch/fletch/out/DebugIA32ClangAsan; export TEMP_DIR="${TMPDIR}";../../out/DebugIA32ClangAsan/natives_to_json ../../out/DebugIA32ClangAsan/natives.json
dyld: Library not loaded: @executable_path/libclang_rt.asan_osx_dynamic.dylib
  Referenced from: /Users/zerny/src/fletch/fletch/src/shared/../../out/DebugIA32ClangAsan/natives_to_json
  Reason: image not found
/bin/sh: line 1: 28725 Trace/BPT trap: 5       ../../out/DebugIA32ClangAsan/natives_to_json ../../out/DebugIA32ClangAsan/natives.json
```

zerny, 2015-08-12:
It looks like fletch_vm_library_generator should also depend on copy_asan

peter-ahe-google, 2015-09-02:
Proposed fix https://codereview.chromium.org/1321353006

commit bd818696 (2015-09-03): Attempt to fix copy_asan dependency issue

Fixes #93

R=ager@google.com

Review URL: https://codereview.chromium.org/1321353006 .
closed 2015-09-03 by peter-ahe-google