dart-archive/sdk #94: unsorted/messaging_process_die_early_gc_on_delete_test may crash the fletch-vm

peter-ahe-google, 2015-08-13:
```
FAILED: fletchc-fletchvm debug_ia32_clang unsorted/messaging_process_die_early_gc_on_delete_test
Expected: Pass Slow 
Actual: Fail
Unexpected compile-time error.
CommandOutput[vm]:
CommandOutput[vm]:

stderr:
fletch-vm(22695,0xb0081000) malloc: *** mach_vm_map(size=131072) failed (error code=3)
*** error: can't allocate region
*** set a breakpoint in malloc_error_break to debug
../../src/vm/object_memory.cc:104: error: Failed to allocate memory of size 1044

Current stacktrace:
  0   fletch-vm                           0x00822f93 _ZN6fletch22DynamicAssertionHelper14DumpStacktraceEv + 51
  1   fletch-vm                           0x00822f12 _ZN6fletch22DynamicAssertionHelper4FailEPKcz + 130
  2   fletch-vm                           0x00802d1e _ZN6fletch5Space18AllocateInNewChunkEi + 174
  3   fletch-vm                           0x00802e0f _ZN6fletch5Space8AllocateEi + 223
  4   fletch-vm                           0x00800c52 _ZN6fletch10HeapObject14CloneInToSpaceEPNS_5SpaceE + 146
  5   fletch-vm                           0x00808212 _ZN6fletch29ScavengeAndChainStacksVisitor28ScavengePointerAndChainStackEPPNS_6ObjectE + 82
  6   fletch-vm                           0x008081ac _ZN6fletch29ScavengeAndChainStacksVisitor10VisitBlockEPPNS_6ObjectES3_ + 44
  7   fletch-vm                           0x0081ddc0 _ZN6fletch14PointerVisitor5VisitEPPNS_6ObjectE + 32
  8   fletch-vm                           0x008058d7 _ZN6fletch7Process35CollectMutableGarbageAndChainStacksEv + 151
  9   fletch-vm                           0x0080954f _ZN6fletch7Program16PrepareProgramGCEb + 111
  10  fletch-vm                           0x00808e40 _ZN6fletch7Program14CollectGarbageEv + 112
  11  fletch-vm                           0x007f5247 _ZN6fletch8GCThread8MainLoopEv + 279
  12  fletch-vm                           0x007f5121 _ZN6fletch8GCThread18GCThreadEntryPointEPv + 17
  13  libsystem_pthread.dylib             0x9686ae13 _pthread_body + 138
  14  libsystem_pthread.dylib             0x9686ad89 _pthread_body + 0
  15  libsystem_pthread.dylib             0x96868e52 thread_start + 34
```

peter-ahe-google, 2015-08-13:
@madsager

mkustermann, 2015-08-19:
So the root cause of this is that

* after a GC the new allocation budget was at least 512 KB
* after a GC the new chunk size was at least 128 KB

This behavior has been in place for a long time.

What this means is that even if a process uses only a 100 bytes, we will blow that to 100s of KB after a mutable collection.

The test uses 5000 processes, which means that it can easily consume `600 MB`. Depending on how much RAM a machine has and how many other tests are running at the same time, this can cause an allocation failure.

The following CL should address it and brings the top memory usage down to roughly `40-60 MB`:
https://codereview.chromium.org/1290793003.

closed 2015-08-19 by mkustermann