dart-archive/sdk #100: Developer friendliness on the command line

ErikCorryGoogle, 2015-08-18:
These should work:
./out/ReleaseIA32Clang/fletch --help
./out/ReleaseIA32Clang/fletch --usage
./out/ReleaseIA32Clang/fletch --version
If you use a name that's not there it should produce the output of help all instead of a stack trace.

The help output for compile-and-run should describe the options instead of just writing [options].
If you use an option wrong it should either describe the correct options or describe a command or URL that describes the correct options.

Most of the help texts are incomprehensible.  eg.
   show      List things or show information about a thing

the show command itself says:

Can't perform 'show' without a target. Try adding a thing or group of things.

I wonder what these things are.  I try one at random:

./out/ReleaseIA32Clang/fletch show thing

It responds with a stack trace.

   compile-and-run [options] dartfile
             Compile and run dartfile in a temporary session.  This is a
             provisionary feature that will be removed shortly

I wonder why it will be removed, and what I can do instead.  Should I feel bad for using it in the mean time?

When I say grep --help, the output starts with:
  Usage: grep [OPTION]... PATTERN [FILE]...
  Search for PATTERN in each FILE or standard input.
  PATTERN is, by default, a basic regular expression (BRE).
  Example: grep -i 'hello world' menu.h main.c

This contains a usage line, a line that explains what grep is useful for, and an example, all in the first 4 lines.  Very nice.  The last few lines are links to more documentation.

peter-ahe-google, 2015-09-16:
@lukechurch any ideas for a one line description of the `fletch` command?

lukechurch, 2015-09-19:
@peter-ahe-google I think we'll need a few more lines than one. To get started how about something like

```
The fletch command line tool manages interactions with the fletch compiler and runtime. 
Example: fletch run sample.dart
To see a list of all available commands use 'fletch help all'
```

peter-ahe-google, 2015-10-01:
Proposed fix [CL 1381083002](https://codereview.chromium.org/1381083002/)

commit 4f78d433 (2015-10-02): Support options in all verbs

Fixes #100

R=lukechurch@google.com, wibling@google.com

Review URL: https://codereview.chromium.org/1381083002 .
closed 2015-10-02 by peter-ahe-google