dart-archive/sdk #99: Running reproduction commands from ./tools/test.py sometimes fail

mkustermann, 2015-08-18:
Here is an example of a failure with a reproduction command:

```
FAILED: none-fletchc debug_ia32_clang_asan unsorted/fletchc_dart2js_vm_test
Expected: Pass 
Actual: CompileTimeError

Command[fletch_session]: tools/testing/bin/linux/dart -c -Dtest
.fletch_session_command.is_incremental_enabled=false ...
```

Running that command resulted in (I've stripped the absolute path prefix for readability):
```
fletch % tools/testing/bin/linux/dart -c -Dtest.fletch_session_command.is_incremental_enabled=false tools/testing/dart/fletch_session_command.dart out/DebugIA32ClangAsan/fletch  tests/unsorted/fletchc_dart2js_test.dart tests/unsorted/fletchc_dart2js_test.dart
     Previous commands in this sesssion:

Running out/DebugIA32ClangAsan/fletch create session 0
$ out/DebugIA32ClangAsan/fletch create session 0
$ out/DebugIA32ClangAsan/fletch create session 0
Can't create session named '0'; There already is a session named '0'.

 => 254
Non-zero exit code (254) from: out/DebugIA32ClangAsan/fletch create session 0
Running out/DebugIA32ClangAsan/fletch x-end session 0
$ out/DebugIA32ClangAsan/fletch x-end session 0
$ out/DebugIA32ClangAsan/fletch x-end session 0
The null object does not have a method 'onData'.

NoSuchMethodError: method not found: 'onData'
Receiver: null
Arguments: [Closure: (dynamic) => dynamic]
#0      Object._noSuchMethod (dart:core-patch/object_patch.dart:42)
#1      Object.noSuchMethod (dart:core-patch/object_patch.dart:45)
#2      IsolateController.attachClient.<attachClient_async_body> (package:fletchc/src/driver/driver_main.dart:452:23)
#3      Future.Future.microtask.<anonymous closure> (dart:async/future.dart:144)
#4      _rootRun (dart:async/zone.dart:900)
#5      _CustomZone.run (dart:async/zone.dart:803)
#6      _CustomZone.runGuarded (dart:async/zone.dart:709)
#7      _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#8      _rootRun (dart:async/zone.dart:904)
#9      _CustomZone.run (dart:async/zone.dart:803)
#10     _CustomZone.runGuarded (dart:async/zone.dart:709)
#11     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#12     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#13     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#14     _Timer._runTimers (dart:isolate-patch/timer_impl.dart:394)
#15     _Timer._handleMessage (dart:isolate-patch/timer_impl.dart:414)
#16     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:148)

 => 253
Non-zero exit code (253) from: out/DebugIA32ClangAsan/fletch x-end session 0
Test outcome: CompileTimeError

```

mkustermann, 2015-08-18:
@peter-ahe-google

peter-ahe-google, 2015-08-18:
The problem is that the persistent process needs to be restarted. Not sure what the best course of action is? I could start by clearly stating that in an error message, and also not including "fletch create session 0" in the try/finally block that runs "fletch x-end session 0".

mkustermann, 2015-08-18:
Personally I ran into this issue now more than once and it disturbs me every time (since I get confused if that is a real issue, or just the re-running not working).

Without knowing how this is even implemented, I can just give some general comments:

a) If ./tools/test.py killed the persistent process before running tests, it could - for symmetry reasons - also kill it before exiting. This would get partially rid of this issue, since the reproduction command will hopefully then just start another driver.

b) The name of the session does not need to be '0' when running the reproduction command. In fact the name doesn't matter. To get a unique name, one could use a uuid or the current timestamp as the session name, thereby avoiding the name conflict.

As a side note: Is it possible to reuse session ids? It seems strange to me, that after ./tools/test.py is done, there is used session left. Why is that not removed?

peter-ahe-google, 2015-08-18:
> As a side note: Is it possible to reuse session ids? It seems strange to me, that after ./tools/test.py is done, there is used session left. Why is that not removed?

Sometimes, the session gets really broken and it cannot be removed. This is a bug, but its hard to anticipate all possible error scenarios: we keep breaking stuff in new ways I haven't seen before. So I'm always interested in a git snapshot of your code with bugs so I can harden the infrastructure against those problems.

peter-ahe-google, 2015-09-07:
See also issue #110

peter-ahe-google, 2015-09-16:
Implementing Martin's suggestion for using the current time to avoid clashes [CL 1346253002](https://codereview.chromium.org/1346253002/). I believe this fixes the original issue reported by Martin, but there's still more to be done, so I wont close this just yet.

peter-ahe-google, 2015-09-18:
I mentioned that I should not include "fletch create session 0" in the try/finally block that runs "fletch x-end session 0". This is addressed as a side-effect of removing x-run [CL 1348673003](https://codereview.chromium.org/1348673003/).

I don't have a reproduction for "NoSuchMethodError: method not found: 'onData'", but fixing issue #92 may solve it.

I have no further action to take on this bug.

peter-ahe-google, 2015-09-25:
I have a repro locally.

peter-ahe-google, 2015-09-28:
I think this is fixed by the fix for #101.

closed 2015-09-28 by peter-ahe-google