dart-archive/sdk #104: Fletch default help-page is unhelpful.

sigurdm, 2015-08-20:
```
$ out/DebugIA32Clang/fletch 
   help      Display this information.
             Use 'fletch help all' for a list of all actions

   show      List things or show information about a thing
```
As I understand it `show` does not work at all - and the description is too general to really understand what it is supposed to do.

madsager, 2015-08-20:
I agree, I think we should remove the show verb for now. It is unimplemented and as far as I can tell nobody is working on it.

peter-ahe-google, 2015-09-07:
I agree, I want to get rid of all prototypes.

ErikCorryGoogle, 2015-09-07:
I would support just printing the output of help all instead of requiring an extra command.

peter-ahe-google, 2015-09-07:
I think it is important that we limit the initial help message, but hopefully @lukechurch can investigate what is more helpful in user studies.

peter-ahe-google, 2015-09-07:
There's precedence both from git and gcc, in limiting the amount of help text shown in the initial help page.

ErikCorryGoogle, 2015-09-07:
They both limit it to about the size of the output of "fletch help all"

erikcorry@bbc:~/fletch-repo/fletch$ ./out/ReleaseIA32/fletch help all | wc
     90     303    2415
erikcorry@bbc:~/fletch-repo/fletch$ gcc --help | wc
     60     428    3905
erikcorry@bbc:~/fletch-repo/fletch$ git help | wc
     42     282    2050

peter-ahe-google, 2015-09-16:
`show` verb removed in [CL 1346223003](https://codereview.chromium.org/1346223003/).

peter-ahe-google, 2015-09-16:
@ErikCorryGoogle the current size of the output of `fletch help all` isn't complete and I expect that it will grow substantially over the next few months. Regardless, it is already twice as big as the output of `git --help`. Between versions 2.2.1 and 2.6.0.rc0..., the output from `git --help` has grown by 30%. I consider this an issue as it doesn't scale, eventually you dump so much information on the user that they don't know where to start. Perhaps 

I want the default help message to be concise and fit within a default terminal window, because I don't want to flood the user with information that they don't need.

That being said, the current text is really bad.

peter-ahe-google, 2015-09-16:
Proposed fix in [CL 1346313003](https://codereview.chromium.org/1346313003/)

commit 0a80ff0c (2015-09-18): Improve 'fletch help' and 'fletch help all'

Fixes #104
Fixes #118
Fixes #119

R=lukechurch@google.com

Review URL: https://codereview.chromium.org/1346313003 .
closed 2015-09-18 by peter-ahe-google