dart-archive/sdk #105: Fletch driver could use a compile-in-fresh-compiler command

sigurdm, 2015-08-20:
For developing on the fletchc compiler it is frustrating to have to kill the fletch resident compiler before running a new compilation.

There could also be a seperate fletchc entrypoint.

madsager, 2015-08-20:
There was, it was removed.

fletch shutdown is your friend. Easy to script.

madsager, 2015-08-20:
Sorry, didn't mean to close the issue; reopened.

Instead of a separate entry point, maybe we can add a convenience script to the repo for the compiler devs if this is slowing you down?

sigurdm, 2015-08-20:
I had not seen the shutdown command. That makes it easier.
I would not say that it slows me down now that I have my own script - but it will slow down anyone else who has to discover it.

sigurdm, 2015-08-20:
What is the difference between `fletch shutdown` and `tools/persistent_process_info.sh -kill` (if any)?

peter-ahe-google, 2015-09-07:
Duplicate of issue #122. The proposed solution is to create a test that is easy to run from the command line as well.

closed 2015-09-07 by peter-ahe-google