dart-archive/sdk #108: VM relying on (tc)malloc implementation can cause 95+% wasted memory

mkustermann, 2015-08-24:
The following simple example:

```
import 'dart:fletch';

main() {
  while (true) {
    var c = new Channel();
    final port = new Port(c);
    Process.spawn(() => port.send(null));
    c.receive();
  }
}
```

Uses after a few seconds of running up to **25 MB** of RAM using tcmalloc. The following are malloc statistics:
```
------------------------------------------------
MALLOC:         177968 (    0.2 MiB) Bytes in use by application
MALLOC: +      4325376 (    4.1 MiB) Bytes in page heap freelist
MALLOC: +      5150032 (    4.9 MiB) Bytes in central cache freelist
MALLOC: +      1099776 (    1.0 MiB) Bytes in transfer cache freelist
MALLOC: +     15461248 (   14.7 MiB) Bytes in thread cache freelists
MALLOC: +       727268 (    0.7 MiB) Bytes in malloc metadata
MALLOC:   ------------
MALLOC: =     26941668 (   25.7 MiB) Actual memory used (physical + swap)
MALLOC: +            0 (    0.0 MiB) Bytes released to OS (aka unmapped)
MALLOC:   ------------
MALLOC: =     26941668 (   25.7 MiB) Virtual address space used
MALLOC:
MALLOC:           1255              Spans in use
MALLOC:             11              Thread heaps in use
MALLOC:           8192              Tcmalloc page size
------------------------------------------------
```

As can be seen, almost all memory consumed by the (unix) process is due to large freelists (in particular in thread local caches).

Doing the same experiment without `-ltcmalloc` (i.e. using glibc's malloc implementation) results in much smaller (unix) process memory usage (between **1 MB and 3 MB**). But even with glibc's malloc, more than 90% of memory is in free lists. Here are glibc's malloc statistics

```
Arena 0:
system bytes     =     135168
in use bytes     =      69880
Arena 1:
system bytes     =     245760
in use bytes     =       5352
Arena 2:
system bytes     =     270336
in use bytes     =      25968
Arena 3:
system bytes     =     249856
in use bytes     =       7600
Arena 4:
system bytes     =     229376
in use bytes     =      31392
Arena 5:
system bytes     =     204800
in use bytes     =       5352
Arena 6:
system bytes     =     159744
in use bytes     =      17048
Arena 7:
system bytes     =     217088
in use bytes     =      46488
Arena 8:
system bytes     =     237568
in use bytes     =       5704
Arena 9:
system bytes     =     188416
in use bytes     =       1248
Arena 10:
system bytes     =     135168
in use bytes     =      17568
Arena 11:
system bytes     =     204800
in use bytes     =       5528
Arena 12:
system bytes     =     258048
in use bytes     =       5352
Arena 13:
system bytes     =     237568
in use bytes     =      54064
Total (incl. mmap):
system bytes     =    2973696
in use bytes     =     298544
max mmap regions =          0
max mmap bytes   =          0
```

On a mobile device it matters a lot whether a process uses `25 MB` of memory or if it's actually free, but it doesn't give it back to the OS.

We could therefore think about:
  - switching to native malloc() implementation
  - having a custom allocator for fixed size objects we allocate/free a lot (e.g. `Process`, `Port`, ... objects).

cc: @madsager @ErikCorryGoogle

madsager, 2015-08-24:
I've been thinking about this as well. Linking in tcmalloc also has a pretty high cost when it comes to binary size. I'm leaning towards getting rid of tcmalloc and doing rolling our own.

skabet, 2015-11-02:
We no longer link with tcmalloc.

closed 2015-11-02 by skabet