dart-archive/sdk #110: Persistent driver crashed / something went wrong during ./tools/test.py

mkustermann, 2015-08-25:
The following happened locally today (does not reproduce). 132 tests worked fine and then one of them failed with:

```
out/DebugIA32ClangAsan/fletch: lost connection to persistent process: Connection reset by peer
```

which caused many others to fail with
```
stderr:
$ out/DebugIA32ClangAsan/fletch compile file /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/io/http_test.dart in session 11
out/DebugIA32ClangAsan/fletch: lost connection to persistent process: Connection reset by peer
Non-zero exit code (255) from: out/DebugIA32ClangAsan/fletch compile file /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/io/http_test.dart in 
session 11
$ out/DebugIA32ClangAsan/fletch x-end session 11
No session named: '11'. Try running 'fletch create session 11'
Non-zero exit code (254) from: out/DebugIA32ClangAsan/fletch x-end session 11
```

See here:

```
$ ./tools/test.py -mdebug --clang --asan
Running: gclient runhooks
Use --run-gclient-hooks=0 to skip this step.

________ running 'download_from_google_storage --no_auth --no_resume --bucket dart-dependencies-fletch -d -r --auto_platform fletch/third_party/libs' in '/usr/local/google
/home/kustermann/repositories/fletch-gclient2'

________ running 'download_from_google_storage --no_auth --no_resume --bucket dart-dependencies-fletch -d -r --auto_platform fletch/third_party/bin' in '/usr/local/google/
home/kustermann/repositories/fletch-gclient2'

________ running 'download_from_google_storage --no_auth --no_resume --bucket dart-dependencies-fletch -d -r --auto_platform fletch/tools/testing/bin' in '/usr/local/googl
e/home/kustermann/repositories/fletch-gclient2'

________ running '/usr/bin/python fletch/tools/clang_update.py --revision=239765' in '/usr/local/google/home/kustermann/repositories/fletch-gclient2'
[linux] Updating Clang to 239765-1 ...
Already up to date.
[mac] Updating Clang to 239765-1 ...
Already up to date.

________ running 'ninja -C fletch' in '/usr/local/google/home/kustermann/repositories/fletch-gclient2'
ninja: Entering directory `fletch'
ninja: no work to do.
Test configurations: none_fletch_warnings_debug_ia32_clang_asan none_fletchd_debug_ia32_clang_asan none_fletch_tests_debug_ia32_clang_asan none_fletchc_debug_ia32_clang_as
an none_fletch_cc_tests_debug_ia32_clang_asan
Killing 26013: out/DebugIA32/dart -c -p /usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/package/ package:fletchc/src/driver/driver_main.dart /usr/loc
al/google/home/kustermann/repositories/fletch-gclient2/fletch/.fletch
Use --kill-persistent-process=0 to skip this step.
Building in out/DebugIA32ClangAsan.
Use --build-before-testing=0 to skip this step.
Building done.
Debug print from fletch_tests works.
[00:08 |  --% | +   29 | -    0]Total: 4448 tests
 * 1 tests will be skipped (1 skipped by design)
 * 0 tests are expected to be flaky but not crash
 * 3420 tests are expected to pass
 * 3 tests are expected to fail that we won't fix
 * 1008 tests are expected to fail that we should fix
 * 6 tests are expected to crash that we should fix
 * 1 tests are allowed to timeout
 * 0 tests are skipped on browsers due to compile-time error

[00:26 |   2% | +  132 | -    0]
FAILED: none-fletch_tests debug_ia32_clang_asan fletch_tests/service_tests/performance
Expected: Pass 
Actual: Fail
CommandOutput[fletch_test]:

stdout:
Running:  tools/cxx_wrapper.py -DFLETCH_CLANG -DFLETCH_ASAN -L/FLETCH_ASAN -std=c++11 -m32 -DFLETCH32 -DFLETCH_TARGET_IA32 -DFLETCH_TARGET_OS_LINUX -DFLETCH_TARGET_OS_POSI
X -I. -Lthird_party/libs/linux/x86 -o out/DebugIA32ClangAsan/generated_service_tests/performance/service_performance_test tests/service_tests/performance/performance_test.
cc tests/service_tests/performance/cc/performance_service.cc tests/service_tests/performance/cc/struct.cc tests/service_tests/performance/cc/unicode.cc out/DebugIA32ClangA
san/libfletch.a -ltcmalloc_minimal -lpthread -ldl -rdynamic
stdout:

stderr:

Running:  out/DebugIA32ClangAsan/fletch compile-and-run tests/service_tests/performance/performance_service_impl.dart -o out/DebugIA32ClangAsan/generated_service_tests/per
formance/performance.snapshot
stdout:

stderr:
out/DebugIA32ClangAsan/fletch: lost connection to persistent process: Connection reset by peer

stderr:
Expect.equals(expected: <0>, actual: <255>) fails.
#0      Expect._fail (package:expect/expect.dart:385:5)
#1      Expect.equals (package:expect/expect.dart:102:5)
#2      Rule.runCommand.<runCommand_async_body> (file:///usr/local/google/home/kustermann/repositories/fletch-gclient2/fletch/tests/service_tests/service_tests.dart:132:12
)
#3      _rootRunUnary (dart:async/zone.dart:911)
#4      _CustomZone.runUnary (dart:async/zone.dart:811)
#5      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#6      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#7      _Future._completeWithValue (dart:async/future_impl.dart:368)
#8      Future.wait.<anonymous closure> (dart:async/future.dart:299)
#9      _rootRunUnary (dart:async/zone.dart:911)
#10     _CustomZone.runUnary (dart:async/zone.dart:811)
#11     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#12     _Future._propagateToListeners (dart:async/future_impl.dart:577)
#13     _Future._completeWithValue (dart:async/future_impl.dart:368)
#14     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#15     _rootRun (dart:async/zone.dart:904)
#16     _CustomZone.run (dart:async/zone.dart:803)
#17     _CustomZone.runGuarded (dart:async/zone.dart:709)
#18     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#19     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#20     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#21     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#22     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

Command[fletch_test]: FletchTestCommand(service_tests/performance)
Took 0:00:06.885363

Short reproduction command (experimental):
    python tools/test.py --clang --asan -rfletch_tests -t60 fletch_tests/service_tests/performance
```

/cc @peter-ahe-google

peter-ahe-google, 2015-09-07:
I think Anders already fixed the problem in the service API. However, it should be easy to reproduce this problem by modifying the assert and then ensure that it doesn't cause other tests to fail.

peter-ahe-google, 2015-09-18:
Upon closer examination, I don't have enough information to reproduce this problem. The expectation (not assertion as I said above) fails *after* the persistent process has closed the socket.

The exit code 255 comes from the `Die` method in the C++ client.

I can reproduce something similar if I `kill -9` the persistent process while running test.py, or if I call exit after writing a snapshot.

One suspicion is that the sesson code called exit after encountering a fletch vm bug during snapshotting. However, this bug was reported on Aug 24 and by then @madsager should have removed all the calls to exit.

I don't think there's anything else I can do in regards to this bug.

closed 2015-09-18 by peter-ahe-google