dart-archive/sdk #111: Flaky test occasionally brakes the build

smanilov, 2015-08-25:
The `service_tests/performance` test occasionally fails unexpectedly and breaks the Fletch build.

Here is the log info from the build system:

```
FAILED: none-fletch_tests release_x64_clang fletch_tests/service_tests/performance
Expected: Pass 
Actual: Fail
CommandOutput[fletch_test]:

stdout:
Running:  tools/cxx_wrapper.py -DFLETCH_CLANG -std=c++11 -DFLETCH_ENABLE_FFI -DFLETCH_ENABLE_LIVE_CODING -DFLETCH_ENABLE_PRINT_INTERCEPTORS -m64 -DFLETCH64 -DFLETCH_TARGET_X64 -DFLETCH_TARGET_OS_MACOS -DFLETCH_TARGET_OS_POSIX -framework CoreFoundation -I. -Lthird_party/libs/mac/x64 -o out/ReleaseX64Clang/generated_service_tests/performance/service_performance_test tests/service_tests/performance/performance_test.cc tests/service_tests/performance/cc/performance_service.cc tests/service_tests/performance/cc/struct.cc tests/service_tests/performance/cc/unicode.cc out/ReleaseX64Clang/libfletch.a -ltcmalloc_minimal -lpthread -ldl -rdynamic
stdout:

stderr:

Running:  out/ReleaseX64Clang/fletch compile-and-run tests/service_tests/performance/performance_service_impl.dart -o out/ReleaseX64Clang/generated_service_tests/performance/performance.snapshot
stdout:

stderr:

Running:  out/ReleaseX64Clang/generated_service_tests/performance/service_performance_test out/ReleaseX64Clang/generated_service_tests/performance/performance.snapshot
stdout:

stderr:

stderr:
Expect.equals(expected: <0>, actual: <-11>) fails.
#0      Expect._fail (package:expect/expect.dart:385:5)
#1      Expect.equals (package:expect/expect.dart:102:5)
#2      Rule.runCommand.<runCommand_async_body> (file:///b/build/slave/fletch-mac-release-x86/build/fletch/tests/service_tests/service_tests.dart:132:12)
#3      _rootRunUnary (dart:async/zone.dart:911)
#4      _CustomZone.runUnary (dart:async/zone.dart:811)
#5      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#6      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#7      _Future._completeWithValue (dart:async/future_impl.dart:368)
#8      Future.wait.<anonymous closure> (dart:async/future.dart:299)
#9      _rootRunUnary (dart:async/zone.dart:911)
#10     _CustomZone.runUnary (dart:async/zone.dart:811)
#11     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#12     _Future._propagateToListeners (dart:async/future_impl.dart:577)
#13     _Future._completeWithValue (dart:async/future_impl.dart:368)
#14     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#15     _rootRun (dart:async/zone.dart:904)
#16     _CustomZone.run (dart:async/zone.dart:803)
#17     _CustomZone.runGuarded (dart:async/zone.dart:709)
#18     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#19     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#20     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#21     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#22     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

Command[fletch_test]: FletchTestCommand(service_tests/performance)
Took 0:00:03.755610

Short reproduction command (experimental):
    python tools/test.py -mrelease -ax64 --kill-persistent-process=0 --run-gclient-hooks=0 --build-before-testing=0 -rfletch_tests -t60 fletch_tests/service_tests/performance
```

peter-ahe-google, 2015-08-26:
Failed on IA32 ASAN. There's more information here:
```
FAILED: none-fletch_tests debug_ia32_clang_asan fletch_tests/service_tests/performance
Expected: Pass 
Actual: Fail
CommandOutput[fletch_test]:

stdout:
Running:  tools/cxx_wrapper.py -DFLETCH_CLANG -DFLETCH_ASAN -L/FLETCH_ASAN -std=c++11 -DFLETCH_ENABLE_FFI -DFLETCH_ENABLE_LIVE_CODING -DFLETCH_ENABLE_PRINT_INTERCEPTORS -m32 -DFLETCH32 -DFLETCH_TARGET_IA32 -DFLETCH_TARGET_OS_MACOS -DFLETCH_TARGET_OS_POSIX -framework CoreFoundation -I. -Lthird_party/libs/mac/x86 -o out/DebugIA32ClangAsan/generated_service_tests/performance/service_performance_test tests/service_tests/performance/performance_test.cc tests/service_tests/performance/cc/performance_service.cc tests/service_tests/performance/cc/struct.cc tests/service_tests/performance/cc/unicode.cc out/DebugIA32ClangAsan/libfletch.a -ltcmalloc_minimal -lpthread -ldl -rdynamic
stdout:

stderr:
ld: warning: PIE disabled. Absolute addressing (perhaps -mdynamic-no-pic) not allowed in code signed PIE, but used in _InterpretFast from out/DebugIA32ClangAsan/libfletch.a(fletch_vm_library.generated.o). To fix this warning, don't compile with -mdynamic-no-pic or link with -Wl,-no_pie

Running:  out/DebugIA32ClangAsan/fletch compile-and-run tests/service_tests/performance/performance_service_impl.dart -o out/DebugIA32ClangAsan/generated_service_tests/performance/performance.snapshot
stdout:

stderr:

Running: DYLD_LIBRARY_PATH=out/DebugIA32ClangAsan out/DebugIA32ClangAsan/generated_service_tests/performance/service_performance_test out/DebugIA32ClangAsan/generated_service_tests/performance/performance.snapshot
stdout:
Sync call took 45.84 us.
    - 21814.63 calls/s
Async call took 13.95 us.
    - 71689.73 calls/s
Building (C++) took 60 us.
Counting (Dart) took 517 us.
Building (Dart) took 391 us.
Counting (C++) took 19 us.

stderr:
ASAN:SIGSEGV
=================================================================
==66474==ERROR: AddressSanitizer: SEGV on unknown address 0xb103edd8 (pc 0x00093d69 bp 0xb01849e8 sp 0xb01849d0 T3)
    #0 0x93d68 in fletch::ThreadIdentifier::IsSelf() const thread_posix.h:28
    #1 0x78e5f in fletch::Service::NotifyResult(fletch::ServiceRequest*) service_api_impl.cc:91
    #2 0x78d89 in PostResultToService service_api_impl.cc:67
    #3 0x15e86 in Native_ForeignVCall1 ffi.cc:476
    #4 0xd17c in BC_InvokeNative generated.S:675

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV thread_posix.h:28 in fletch::ThreadIdentifier::IsSelf() const
Thread T3 created by T1 here:
    #0 0x1a53f3  (libclang_rt.asan_osx_dynamic.dylib+0x433f3)
    #1 0x93b81 in fletch::Thread::Run(void* (*)(void*), void*) thread_posix.cc:24
    #2 0x92a22 in fletch::ThreadPool::TryStartThread(void (*)(void*), void*, int) thread_pool.cc:49
    #3 0x658b7 in fletch::Scheduler::Run() scheduler.cc:267
    #4 0x1831c in fletch::RunSnapshot(fletch::List<unsigned char>) fletch_api_impl.cc:42
    #5 0x185e7 in fletch::RunShapshotFromFile(char const*) fletch_api_impl.cc:55
    #6 0x35c1 in DartThreadEntry(void*) (service_performance_test+0x35c1)
    #7 0x1be43d  (libclang_rt.asan_osx_dynamic.dylib+0x5c43d)

Thread T1 created by T0 here:
    #0 0x1a53f3  (libclang_rt.asan_osx_dynamic.dylib+0x433f3)
    #1 0x34f6 in RunSnapshotInNewThread(char*) (service_performance_test+0x34f6)
    #2 0x2751 in SetupPerformanceTest(int, char**) (service_performance_test+0x2751)
    #3 0x2677 in main (service_performance_test+0x2677)
    #4 0x98779700 in start (libdyld.dylib+0x3700)
    #5 0x1  (service_performance_test)+0x1)

==66474==ABORTING

stderr:
Expect.equals(expected: <0>, actual: <-6>) fails.
#0      Expect._fail (package:expect/expect.dart:385:5)
#1      Expect.equals (package:expect/expect.dart:102:5)
#2      Rule.runCommand.<runCommand_async_body> (file:///Volumes/data/b/build/slave/fletch-mac-asan-x86/build/fletch/tests/service_tests/service_tests.dart:132:12)
#3      _rootRunUnary (dart:async/zone.dart:911)
#4      _CustomZone.runUnary (dart:async/zone.dart:811)
#5      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#6      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#7      _Future._completeWithValue (dart:async/future_impl.dart:368)
#8      Future.wait.<anonymous closure> (dart:async/future.dart:299)
#9      _rootRunUnary (dart:async/zone.dart:911)
#10     _CustomZone.runUnary (dart:async/zone.dart:811)
#11     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#12     _Future._propagateToListeners (dart:async/future_impl.dart:577)
#13     _Future._completeWithValue (dart:async/future_impl.dart:368)
#14     _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#15     _rootRun (dart:async/zone.dart:904)
#16     _CustomZone.run (dart:async/zone.dart:803)
#17     _CustomZone.runGuarded (dart:async/zone.dart:709)
#18     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:734)
#19     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#20     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#21     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#22     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

Command[fletch_test]: FletchTestCommand(service_tests/performance)
Took 0:00:05.601704

Short reproduction command (experimental):
    python tools/test.py --kill-persistent-process=0 --run-gclient-hooks=0 --build-before-testing=0 --asan -rfletch_tests -t60 fletch_tests/service_tests/performance
```

skabet, 2015-09-02:
Proposed fix: https://codereview.chromium.org/1325063003/

closed 2015-09-02 by skabet