dart-archive/sdk #112: Debugger: unable to restart after compile-time error

peter-ahe-google, 2015-08-26:
The incremental feature_test.dart **compile_time_error_hides_field** creates a compile time error, and I think the debugger is unable to restart after that.

Repro:

```
./out/DebugIA32Clang/dart -ppackage -c tests/fletchc/incremental/feature_test.dart  compile_time_error_hides_field
```

Notice that the debugger prints `### process not running` below instead of running the updated program.

```
Test 'compile_time_error_hides_field'
Running 'out/DebugIA32Clang/fletch-vm -Xvalidate-heaps'
Program version 1 #1:
==> main.dart <==
1: // Regression test for what happens when the parser doesn't recover.
2: class A {
3:   // TODO(ahe): should just expect "null"
4:   bool operator ===(A other) { return true; }
5: 
6:   int field;
7: }
8: 
9: main() {
10:   print(new A().field);
11: }

org.dartlang.fletch.test-case:/main.dart:4:17:
Error: Unexpected token '==='.
  bool operator ===(A other) { return true; }
                ^^^
org.dartlang.fletch.test-case:/main.dart:10:9:
Warning: No member named 'field' in class 'A'.
  print(new A().field);
        ^^^^^^^^^^^^^
### 'finish' cancelled because another breakpoint was hit
fletch_vm_stdout: Compile error
Stack trace:
> 0: yield        	
  1: compileError 	@yield(InterruptKind.compileTimeError.index)
  2: A initializer	
  3: main         	@new A()
  4: callMain     	@invokeMain(arguments)
  5: entry        	@callMain([] )

Compile-time error expected
Program version 2 #1:
==> main.dart <==
1: // Regression test for what happens when the parser doesn't recover.
2: class A {
3:   // TODO(ahe): should expect "null"
4: 
5:   int field;
6: }
7: 
8: main() {
9:   print(new A().field);
10: }

Reusing origin library(dart:core) (assumed read-only).
Reusing library(dart:_fletch_system) (assumed read-only).
Reusing origin library(dart:_internal) (assumed read-only).
Reusing origin library(dart:collection) (assumed read-only).
Reusing origin library(dart:math) (assumed read-only).
Reusing library(dart:fletch) (assumed read-only).
Reusing origin library(dart:convert) (assumed read-only).
Reusing origin library(dart:async) (assumed read-only).
Reusing library(dart:fletch.io) (assumed read-only).
Reusing library(dart:fletch.ffi) (assumed read-only).
Reusing origin library(dart:typed_data) (assumed read-only).
Reusing library(dart:system) (assumed read-only).
Attempting to reuse library(org.dartlang.fletch.test-case:/main.dart).
New library synthesized.
Differences computed.
Looking at difference: Modified(class(A) -> class(A))
Simple modification of class(A) detected
Differences computed.
Looking at difference: Added(Instance of 'PartialFieldList')
Add field field(A#field) to class(A).
CommandCode.PrepareForChanges()
CommandCode.RemoveFromMap(MapId.methods, 4)
CommandCode.PushNull()
CommandCode.PushNull()
CommandCode.PushNull()
CommandCode.PushNewFunction(0, 3, [load const @0, invoke static @1, invoke 49408, invoke static @2, pop, load literal null, return 1 0, method end 50], [])
CommandCode.PopToMap(MapId.methods, 202)
CommandCode.PushFromMap(MapId.classes, 0)
CommandCode.PushNewInteger(17409)
CommandCode.PushFromMap(MapId.methods, 85)
CommandCode.PushNewInteger(34817)
CommandCode.PushFromMap(MapId.methods, 195)
CommandCode.PushNewInteger(35840)
CommandCode.PushFromMap(MapId.methods, 194)
CommandCode.PushNewInteger(56320)
CommandCode.PushFromMap(MapId.methods, 192)
CommandCode.PushNewInteger(109569)
CommandCode.PushFromMap(MapId.methods, 196)
CommandCode.PushNewInteger(115968)
CommandCode.PushFromMap(MapId.methods, 0)
CommandCode.ChangeMethodTable(6)
CommandCode.PushFromMap(MapId.classes, 21)
CommandCode.PushNewInteger(49408)
CommandCode.PushFromMap(MapId.methods, 0)
CommandCode.PushNewInteger(49665)
CommandCode.PushFromMap(MapId.methods, 2)
CommandCode.PushNewInteger(145664)
CommandCode.PushFromMap(MapId.methods, 0)
CommandCode.ChangeMethodTable(3)
CommandCode.PushNewString('A()')
CommandCode.PopToMap(MapId.constants, 78)
CommandCode.PushFromMap(MapId.classes, 21)
CommandCode.PushFromMap(MapId.classes, 0)
CommandCode.ChangeSuperClass()
CommandCode.PushFromMap(MapId.methods, 202)
CommandCode.PushFromMap(MapId.constants, 78)
CommandCode.ChangeMethodLiteral(0)
CommandCode.PushFromMap(MapId.methods, 202)
CommandCode.PushFromMap(MapId.methods, 197)
CommandCode.ChangeMethodLiteral(1)
CommandCode.PushFromMap(MapId.methods, 202)
CommandCode.PushFromMap(MapId.methods, 5)
CommandCode.ChangeMethodLiteral(2)
CommandCode.PushFromMap(MapId.classes, 21)
CommandCode.PushNewInteger(0)
CommandCode.PushNull()
CommandCode.PushNewArray(2)
CommandCode.ChangeSchemas(1, 1)
CommandCode.PushFromMap(MapId.methods, 11)
CommandCode.PushFromMap(MapId.methods, 202)
CommandCode.ChangeMethodLiteral(0)
CommandCode.CommitChanges(8)
Stack trace:
> 0: yield        	
  1: compileError 	@yield(InterruptKind.compileTimeError.index)
  2: A initializer	
  3: <missing>    	
  4: callMain     	@invokeMain(arguments)
  5: entry        	@callMain([] )

### process not running
VM exited with exit code: -15.
Test 'compile_time_error_hides_field' failed
Unhandled exception:
Uncaught Error: Bad state: Test has 1 problem(s). Details:
Problem #1:
Expect.listEquals(list length, expected: <1>, actual: <0>) fails: Next element <null>
#0      Expect._fail (package:expect/expect.dart:385:5)
#1      Expect.listEquals (package:expect/expect.dart:202:7)
#2      compileAndRun.<compileAndRun_async_body>.<anonymous closure>.<<anonymous closure>_async_body> (tests/fletchc/incremental/feature_test.dart:267:16)
#3      _RootZone.runUnary (dart:async/zone.dart:1166)
#4      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#5      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#6      _Future._completeWithValue (dart:async/future_impl.dart:368)
#7      _Future._asyncComplete.<anonymous closure> (dart:async/future_impl.dart:422)
#8      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#9      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#10     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#11     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

Stack Trace:
#0      TestSession.waitForCompletion.<waitForCompletion_async_body> (tests/fletchc/incremental/feature_test.dart:494:7)
#1      _RootZone.runUnary (dart:async/zone.dart:1166)
#2      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#3      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#4      _Future._addListener.<anonymous closure> (dart:async/future_impl.dart:274)
#5      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#6      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#7      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#8      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

#0      _rootHandleUncaughtError.<anonymous closure> (dart:async/zone.dart:895)
#1      _microtaskLoop (dart:async/schedule_microtask.dart:43)
#2      _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#3      _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#4      _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)
```

@madsager @wibling @lukechurch

peter-ahe-google, 2015-08-26:
**update_dependencies_recoverable_compile_time_error** is also affected by this.

mkustermann, 2015-08-26:
There is a test, which actually tests that one can restart a frame after a compile time error, namely `tests/debugger/restart_compile_time_error_test.dart`. But that test doesn't test changing the application code and then restarting it.

madsager, 2015-08-26:
The debugger is capable of restarting on a compile time error, so that is not the issue. The issue is that there is a frame on the stack that we do not understand after the rewrite (main).

peter-ahe-google, 2015-08-26:
And **update_dependencies_unrecoverable_compile_time_error**

peter-ahe-google, 2015-08-26:
What about `Session.checkRunning`?

peter-ahe-google, 2015-08-26:
Perhaps I'm supposed to do something else than calling `Session.restart`?

madsager, 2015-08-26:
Peter and I just chatted about this. The test harness currently assumes that we only stop in main. Therefore, it just uses Session.restart to restart main. In this case we are not in main (instead we have a stack and are stopped at a compile-time error) and therefore we need to select the right frame to restart before restarting. So, yeah, I did it wrong when I rewrote the tests to use Session.restart. ;-)

peter-ahe-google, 2015-08-26:
And there's nothing wrong with the debugger, the problem is in how feature_test uses the debugger. Which is actually awesome! (and easy to fix)

peter-ahe-google, 2015-08-26:
Proposed fix here https://codereview.chromium.org/1310733006/

commit e7089a79 (2015-08-28): Select callMain's frame to handle compile-time errors in feature_test.dart

Fixes #112

R=ager@google.com

Review URL: https://codereview.chromium.org/1310733006 .
closed 2015-08-28 by peter-ahe-google