dart-archive/sdk #119: fletch help all is not very helpful

ErikCorryGoogle, 2015-09-01:
eg:

   create    Create something

Doesn't really add much.

peter-ahe-google, 2015-09-16:
From a conversation in [CL 1332563007](https://codereview.chromium.org/1332563007/diff/80001/pkg/fletchc/lib/src/verbs/documentation.dart#newcode126) between me and @lukechurch:

> - What do the [brackets] imply?

By convention, in command line documentation, square brackets indicate an
optional element.

For example, this is the synopsis of the "ls" command from my Mac:

     ls [-ABCFGHLOPRSTUW@abcdefghiklmnopqrstuwx1] [file ...]

From "man ls" aka ls(1).

This is the DIR command from Windows:

     dir [<Drive>:][<Path>][<FileName>] [...] [/p] [/q] [/w] [/d] [/a[[:]<Attributes>]][/o[[:]<SortOrder>]] [/t[[:]<TimeField>]] [/s] [/b] [/l] [/n] [/x] [/c] [/4]

From https://technet.microsoft.com/en-us/library/cc755121.aspx

This is the complete documentation of the syntax used by Microsoft:
https://technet.microsoft.com/en-us/library/cc771080.aspx

The Unix command line syntax is documented by the POSIX standard:

http://pubs.opengroup.org/onlinepubs/9699919799/basedefs/V1_chap12.html

Perhaps we should adopt similar conventions. In my opinion the Microsoft format
is slightly more informative than the POSIX conventions. For example, I think it
is clearer that both "/p" and "/q" are valid options to "dir" whereas I don't
think it is that obvious that "-A" and "-@" are valid options to "ls". It
implies knowledge that options using only one dash often are single letter
options (that may be combined, that is, "ls -A -@" is equivalent to "ls -@ -A",
"ls -A@", and "ls -@A").

On the other hand, whereas the POSIX command line syntax synopsis is less
informative than the Microsoft syntax, most Microsoft tools are really brittle
when it comes to parsing options. For example, it may be required that "/p"
comes before "/q" in the "dir" command. This is due to the lack of a general
library function for parsing command line options in Windows and some really
funky ways to parse quotes on the command line
(http://blogs.msdn.com/b/twistylittlepassagesallalike/archive/2011/04/23/every...).

So in my opinion, no lessons should be learned from Windows when it comes to how
command line arguments are parsed, but the Windows documentation conventions are
a better source of inspiration than POSIX.

peter-ahe-google, 2015-09-16:
Proposed fix in [CL 1346313003](https://codereview.chromium.org/1346313003/)

commit 0a80ff0c (2015-09-18): Improve 'fletch help' and 'fletch help all'

Fixes #104
Fixes #118
Fixes #119

R=lukechurch@google.com

Review URL: https://codereview.chromium.org/1346313003 .
closed 2015-09-18 by peter-ahe-google