dart-archive/sdk #120: "fletch show help" causes uncaught exception and stack trace

ErikCorryGoogle, 2015-09-01:
I would expect an error message instead, perhaps a pointer to the sorts of "things" that "show" can show information about.

peter-ahe-google, 2015-09-16:
Proposed fix in [CL 1348763003](https://codereview.chromium.org/1348763003/):

<pre><code>$ <strong>fletch show help</strong>
   show      List things or show information about a thing

$ <strong>fletch compile help</strong>
   compile file FILE
             Compile file named FILE
</code></pre>

I believe we have another bug tracking the fact that show isn't documented.

peter-ahe-google, 2015-09-16:
Well, I can't find a bug that tracks "show" not being documented, but this CL removes that verb: [CL 1346223003](https://codereview.chromium.org/1346223003).

commit 1ee58cda (2015-09-18): Create pseudo target for verb help

Fixes #120

R=lukechurch@google.com

Review URL: https://codereview.chromium.org/1348763003 .
closed 2015-09-18 by peter-ahe-google