dart-archive/sdk #121: fletch shutdown does not produce any output

ErikCorryGoogle, 2015-09-01:
It might be helpful to print one of the following messages, to reassure the user that it was not a no-op:

VM closed down on localhost:1234
No VM found on localhost:1234

peter-ahe-google, 2015-09-16:
I guess I need to improve the documentation of this command.

This shuts down the persistent process, not a Fletch VM.

That being said, we probably also need a command to detach from a Fletch VM that was previously attached. If that command needs to be called `fletch shutdown`, then we'll just have to find a different name for the current `fletch shutdown`.

@lukechurch any thoughts?

peter-ahe-google, 2015-09-18:
Proposed fix: [CL 1354213003](https://codereview.chromium.org/1354213003/)

commit 74221c55 (2015-09-21): Make shutdown work again

Fixes #121

R=lukechurch@google.com

Review URL: https://codereview.chromium.org/1354213003 .
closed 2015-09-21 by peter-ahe-google