dart-archive/sdk #122: Non-persistent version of "out/*/fletch compile-and-run myfile.dart"

skabet, 2015-09-02:
All my current development issues can be solved by a version of 

  out/*/fletch compile-and-run myfile.dart

that doesn't use a persistent process. Preferable, just a flag:

  out/*/fletch compile-and-run --no-persistent myfile.dart

and with a short version as well :)

peter-ahe-google, 2015-09-02:
Could you give some examples of what problems you have? I think I have an idea, and I'm not sure that "not persistent" is precisely what you need.

skabet, 2015-09-02:
Basically a mode where the no process is detached, and all VM's launched are a child-process of the entry point (may it be several processes down). Makes sense?

peter-ahe-google, 2015-09-02:
Yes. Thank you :-)

peter-ahe-google, 2015-09-03:
We're currently thinking of solving this by having tests that can also be run from the command line. Similar to [feature_test.dart](https://github.com/dart-lang/fletch/blob/a198e667b113a51a5dc486c7be46bf78f06c4905/tests/fletchc/incremental/feature_test.dart), but feature_test.dart should also be generalized to accept files on the command line (for when you don't want to modify [tests_with_expectations.dart](https://github.com/dart-lang/fletch/blob/7d6c8af48e05971fef450a25af5f8367db4b4d9c/tests/fletchc/incremental/tests_with_expectations.dart)).

peter-ahe-google, 2015-09-08:
Proposed fix in [CL 1307203007](https://codereview.chromium.org/1307203007/)

commit 272aa4a6 (2015-09-09): Implement end-to-end test for easy debugging of Fletch VM

Fixes #122

R=ager@google.com

Review URL: https://codereview.chromium.org/1307203007 .
closed 2015-09-09 by peter-ahe-google