dart-archive/sdk #124: Add regression test for issue #14

peter-ahe-google, 2015-09-15:
I'm not sure how to test for issue #14, so I'm filing this bug so we don't forget to test it.

It should probably be part of a sanity test we do on our SDK directory.

peter-ahe-google, 2015-09-15:
@wibling @ricowind Let me know if you have any suggestions.

peter-ahe-google, 2015-09-15:
Also test that the current directory is respected. The current working directory of the persistent process (Uri.base) is where the first fletch program ran. If the user changes directory, files should be read relative to the client, not the persistent process. This is implemented in [CL 1342673004](https://codereview.chromium.org/1342673004), but not tested yet.

mit-mit, 2015-12-07:
@peter-ahe-google sounds like we can close this?

closed 2015-12-07 by mit-mit