dart-archive/sdk #127: ARM cross compilation fails on a Mac

mit-mit, 2015-09-16:
The main compilation works fine on a Mac using the [build instructions](https://github.com/dart-lang/fletch/wiki/Building), but cross compilation to ARM fails.

mit-mit, 2015-09-16:
Build output:

```
ninja: Entering directory `out/DebugXARM'
[1/115] SOLINK libffi_test_library.dylib, POSTBUILDS
[2/115] LINK fletch, POSTBUILDS
[3/115] ACTION(host) natives_json: make_natives_json_ec0dbd6e96ed66ac5a5c54cbd17c698c
[4/115] CXX obj/src/vm/fletch_vm_library_base.event_handler_linux.o
[5/115] CXX obj/src/vm/fletch_vm_library_base.event_handler_lk.o
[6/115] CXX obj/src/vm/fletch_vm_library_base.event_handler_mbed.o
[7/115] CXX obj/src/vm/fletch_vm_library_base.heap.o
FAILED: ../../tools/cxx_wrapper.py -framework CoreFoundation -Wl,-dead_strip -L../../third_party/libs/mac/arm -arch armv7 -L. -o fletch obj/src/tools/driver/fletch.main.o obj/src/tools/driver/fletch.connection.o obj/src/tools/driver/fletch.platform_linux.o obj/src/tools/driver/fletch.platform_macos.o libfletch_shared.a  -lpthread
ld: warning: directory not found for option '-L../../third_party/libs/mac/arm'
ld: library not found for -lcrt1.3.1.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
FAILED: if [ ! -e libffi_test_library.dylib -o ! -e libffi_test_library.dylib.TOC ] || otool -l libffi_test_library.dylib | grep -q LC_REEXPORT_DYLIB ; then ../../tools/cc_wrapper.py -shared -framework CoreFoundation -Wl,-dead_strip -L../../third_party/libs/mac/arm -arch armv7 -L. -install_name /usr/local/lib/libffi_test_library.dylib -o libffi_test_library.dylib @libffi_test_library.dylib.rsp && { otool -l libffi_test_library.dylib | grep LC_ID_DYLIB -A 5; nm -gP libffi_test_library.dylib | cut -f1-2 -d' ' | grep -v U$; true; } > libffi_test_library.dylib.TOC; else ../../tools/cc_wrapper.py -shared -framework CoreFoundation -Wl,-dead_strip -L../../third_party/libs/mac/arm -arch armv7 -L. -install_name /usr/local/lib/libffi_test_library.dylib -o libffi_test_library.dylib @libffi_test_library.dylib.rsp && { otool -l libffi_test_library.dylib | grep LC_ID_DYLIB -A 5; nm -gP libffi_test_library.dylib | cut -f1-2 -d' ' | grep -v U$; true; } > libffi_test_library.dylib.tmp && if ! cmp -s libffi_test_library.dylib.tmp libffi_test_library.dylib.TOC; then mv libffi_test_library.dylib.tmp libffi_test_library.dylib.TOC ; fi; fi
ld: warning: directory not found for option '-L../../third_party/libs/mac/arm'
ld: warning: ignoring file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks//CoreFoundation.framework/CoreFoundation, missing required architecture armv7 in file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks//CoreFoundation.framework/CoreFoundation (2 slices)
ld: warning: ignoring file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/lib/libSystem.dylib, missing required architecture armv7 in file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/lib/libSystem.dylib (2 slices)
Undefined symbols for architecture armv7:
  "_malloc", referenced from:
      _pfun0 in ffi_test_library.ffi_test_library.o
      _pfun1 in ffi_test_library.ffi_test_library.o
      _pfun2 in ffi_test_library.ffi_test_library.o
      _pfun3 in ffi_test_library.ffi_test_library.o
      _pfun4 in ffi_test_library.ffi_test_library.o
      _pfun5 in ffi_test_library.ffi_test_library.o
      _pfun6 in ffi_test_library.ffi_test_library.o
      ...
  "___error", referenced from:
      _ifun0EINTR in ffi_test_library.ffi_test_library.o
      _ifun1EINTR in ffi_test_library.ffi_test_library.o
      _ifun2EINTR in ffi_test_library.ffi_test_library.o
      _ifun3EINTR in ffi_test_library.ffi_test_library.o
      _ifun4EINTR in ffi_test_library.ffi_test_library.o
      _ifun5EINTR in ffi_test_library.ffi_test_library.o
      _ifun6EINTR in ffi_test_library.ffi_test_library.o
      ...
ld: symbol(s) not found for architecture armv7
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

mit-mit, 2015-09-16:
Added a note in the build instructions.

sgjesse, 2015-09-17:
Seems like the ffi tests stopped building for the ARM cross compilation.

mkustermann, 2015-09-17:
That might not be the only thing since it fails very early. Another strange thing: Why is it compiling "_linux.o" *and* "_lk.o" *and* "_mbed.o" ? I assume that the gyp files are not even ready for ARM cross compilation on mac, cc: @peter-ahe-google

peter-ahe-google, 2015-09-17:
@mkustermann, @herhut-ggl had to change source filtering to enable cross-compilation. See [CL 1267563002](https://codereview.chromium.org/1267563002).

mit-mit, 2015-12-07:
By triage discussion, closing this as won't fix.

closed 2015-12-07 by mit-mit