dart-archive/sdk #130: Fletch version and build number

peter-ahe-google, 2015-09-16:
I need the fletch version and build number for implementing `fletch --version` (see #100), and for including in crash reports see [CL 1345213002](https://codereview.chromium.org/1345213002/).

I guess we can use `dart/tools/make_version.py`, but I suggest we run it from `gclient runhooks` instead of during the build.

@mkustermann @ricowind WDYT?

ricowind, 2015-09-17:
The reason for not running it as part of gclient runhooks was that we wanted to make sure (by request) that local builds actually had the correct hash (or svn revision back in the day) in the version string. I don't know if this has actually ever made any difference in the real world. Also, this does not really work since you doing a  local git commit will change the version, but not trigger a rebuild of the files, also, you can't be sure if there are non committed stuff locally. So all in all, I have no strong feelings if we do this at compile or runhooks time - for official builds on the bots the version should always be correct anyway.

mkustermann, 2015-09-17:
I think there are several aspects to think about here. The first question is what the version number should look like (in fact, I raised that question some time ago). If we want to have a semantic version, we need to determine major/minor/patch and postfix. In the dart SDK we have checked in `tools/VERSION` which describes these. Together with `tools/VERSION` the script in `tools/utils.py:GetSemanticSDKVersion()` will build a corresponding version string. That script reaches out to `git` if necessary to get the commit hash of `HEAD`.

The second aspect when to generate the version string and how programs know about it. One option is to built the version into the the programs (e.g. the persistent process). Another option is to put it into a file  which the programs can find (in the Dart sdk different programs behave differently IIRC, some navigate to the version file and read it's contents, some have it baked into the source).

Do we want to have this number always, or just for the finished SDK bundle?

If we do that via `gclient runhooks`, where do we place the version? Would that generate a `.dart` / `.cc` file which all our programs can import/include so they have it baked in?

(Personally I don't run `gclient runhooks` or `gclient sync` very often (using `git fetch`, `git rebase`, `ninja`, ...), so the hash on my builds will always be wrong and thereby meaningless. For the buildbot it doesn't matter, as @ricowind  says.)

peter-ahe-google, 2015-09-17:
Short term solution in [CL 1356603002](https://codereview.chromium.org/1356603002/)

I discussed this with @ErikCorryGoogle and he mentioned that this creates a non-reproducible build, which is probably something we should strive to avoid. So when we more frequent releases, perhaps we can simply rely on a file that we modify each time we push to trunk/stable/what-it's-called.

ricowind, 2015-09-21:
just to be clear,  [CL 1356603002](https://codereview.chromium.org/1356603002/) does not actually add the flag right?

peter-ahe-google, 2015-09-21:
Yes. Issue #100 is about (amongst other things) to add a --version flag to the `fletch` command.

But this issue is fixed by 85fb1ed34f850ce0588d4daaa6a85c6b0a04e896, so I'll close it.

closed 2015-09-21 by peter-ahe-google