dart-archive/sdk #133: fletch-vm segmentation fault

mit-mit, 2015-09-18:
Ran the attached program on the Raspberry Pi 2 and got a segmentation fault.

mit-mit, 2015-09-18:
Commands I ran on the Raspberry:
```
 ./fletch/fletch-vm --port=10000 --host=0.0.0.0 &
```

Output:
```
[1]+  Segmentation fault      ./fletch/fletch-vm --port=10000 --host=0.0.0.0
```

kasperl, 2015-09-18:
Which program was it? Maybe I just can't figure out where to find the attachment.

mit-mit, 2015-09-18:
Commands I ran on the desktop:
```
mit@mit-pc:~/Developer/github/fletch/fletch/out/DebugIA32Clang$ ./fletch create session mit4
Created session 'mit4'.
mit@mit-pc:~/Developer/github/fletch/fletch/out/DebugIA32Clang$ ./fletch attach tcp_socket 192.168.1.1:10000 in session mit4
Connected to Fletch VM on TCP socket 58511 -> 10000
mit@mit-pc:~/Developer/github/fletch/fletch/out/DebugIA32Clang$ ./fletch compile file ~/Developer/dart/pi-led/led.dart in session mit4
Compiled 'file:///usr/local/google/home/mit/Developer/dart/pi-led/led.dart' to 3407 commands

mit@mit-pc:~/Developer/github/fletch/fletch/out/DebugIA32Clang$ ./fletch x-run in session mit4
```

Output:
```
Internal error: No command received from Fletch VM
#0      throwInternalError (package:fletchc/src/diagnostic.dart:165:3)
#1      xRunTask.<xRunTask_async_body> (package:fletchc/src/verbs/x_run_verb.dart:84:5)
#2      _rootRunUnary (dart:async/zone.dart:911)
#3      _CustomZone.runUnary (dart:async/zone.dart:811)
#4      _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#5      _Future._propagateToListeners (dart:async/future_impl.dart:577)
#6      _Future._chainCoreFuture (dart:async/future_impl.dart:341)
#7      _Future._asyncComplete (dart:async/future_impl.dart:407)
#8      _AsyncCompleter.complete (dart:async/future_impl.dart:41)
#9      FletchVmSession.shutdown.<shutdown_async_body> (package:fletchc/session.dart)
#10     _rootRunUnary (dart:async/zone.dart:911)
#11     _CustomZone.runUnary (dart:async/zone.dart:811)
#12     _Future._propagateToListeners.handleValueCallback (dart:async/future_impl.dart:494)
#13     _Future._propagateToListeners (dart:async/future_impl.dart:577)
#14     _Future._addListener.<anonymous closure> (dart:async/future_impl.dart:274)
#15     _rootRun (dart:async/zone.dart:904)
#16     _CustomZone.run (dart:async/zone.dart:803)
#17     _CustomZone.bindCallback.<anonymous closure> (dart:async/zone.dart:736)
#18     _microtaskLoop (dart:async/schedule_microtask.dart:43)
#19     _microtaskLoopEntry (dart:async/schedule_microtask.dart:52)
#20     _runPendingImmediateCallback (dart:isolate-patch/isolate_patch.dart:96)
#21     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:149)

```

mit-mit, 2015-09-18:
Hmm, non picture attachments don't work. Here is the file:
https://drive.google.com/open?id=0B7hRJD_gwzX1b2lnWF9hU3paSDA

closed 2015-11-13 by mit-mit