dart-archive/sdk #134: Control logging output from 'fletch' command

kasperl, 2015-09-18:
Currently, when running the 'fletch' command there's a lot of useful information getting logged:

```
$ fletch run file benchmarks/DeltaBlue.dart 
Compiled 'file:benchmarks/DeltaBlue.dart' to 4022 commands

Connected to Fletch VM on TCP socket 50746 -> 55116
DeltaBlue(RunTime): 51042 us.
```

It would be interesting to hide the logging information by default (unless explicitly requested through some `verbose` or `show logs` option), but it would be even better if we could print the logs automatically in case of hitting an error condition. Something like:

```
$ fletch run benchmarks/DeltaBlue.dart
DeltaBlue(RunTime): 51042 us.
```

and

```
$ fletch run benchmarks/DeltaBlue.dart
[compiled 'file:benchmarks/DeltaBlue.dart' to 4022 commands]
[connected to Fletch VM on TCP socket 50746 -> 55116]
Unexpected error: Fletch VM is simply too fast!
```

peter-ahe-google, 2015-09-18:
Proposed fix: [CL 1343353003](https://codereview.chromium.org/1343353003/)

```console
$ fletch run hello.dart 
Hello, World!
$ fletch run mirrors.dart
/Users/ahe/Dart/fletch-repo/fletch/mirrors.dart:
Error: Fletch doesn't support 'dart:mirrors'. See https://goo.gl/Kwrd0O

/Users/ahe/Dart/fletch-repo/fletch/mirrors.dart:4:9:
Error: 'runtimeType' isn't supported in Fletch. See https://goo.gl/ELH6Zc
  print(new Object().runtimeType);
        ^^^^^^^^^^^^^^^^^^^^^^^^
Compile error
default: [2015-09-18 23:31:31.960] Created session 'default'.
default: [2015-09-18 23:31:33.924] Compiled 'file:///Users/ahe/Dart/fletch-repo/fletch/hello.dart' to 2228 commands
default: [2015-09-18 23:31:33.989] Connected to Fletch VM on TCP socket 56235 -> 56234
default: [2015-09-18 23:31:51.938] Compiled 'file:///Users/ahe/Dart/fletch-repo/fletch/mirrors.dart' to 2258 commands
default: [2015-09-18 23:31:51.946] Connected to Fletch VM on TCP socket 56237 -> 56236
Compile-time error
$
```

kasperl, 2015-09-21:
Looks much better. One question: Where do users of command line tools expect to see the most valuable information? At the beginning or at the end of the output? 

What's the 'Compile error' and 'Compile-time error' lines doing?

peter-ahe-google, 2015-09-21:
> What's the 'Compile error' and 'Compile-time error' lines doing?

The first line is output from the Fletch VM. I don't think it should be printing it. The second line comes from the fletchc code that understands what kind of problem occurred.

I've filed #136 and #137 to track this.

peter-ahe-google, 2015-09-21:
> Looks much better. One question: Where do users of command line tools expect to see the most valuable information? At the beginning or at the end of the output?

I don't know. My guess is that most people look at the end of the output, but experienced developers learn to look at the beginning of the output.

commit c078a739 (2015-09-22): Be less verbose unless there's a problem

Fixes #134

R=kasperl@google.com

Review URL: https://codereview.chromium.org/1343353003 .
closed 2015-09-22 by peter-ahe-google