dart-archive/sdk #135: A bit of type checking would be nice

mit-mit, 2015-09-18:
When running code with fletch, no type checking seems to take place. Is that expected?

Repro steps:

Save the following code to main.dart:
```
import 'package:gpio/gpio.dart';

main(List<String> args) {
  const int led = 47;

  var gpio = new PiMemoryMappedGPIO();
  gpio.setMode(led, Mode.output);
  gpio.setPin(led, 1);
}
```

Start a VM on a Raspberry PI
```
sudo ./fletch/fletch-vm --port=10004 --host=0.0.0.0
```

Run main.dart:
```
$ ./out/DebugIA32Clang/fletch attach tcp_socket 192.168.1.1:10004
Connected to Fletch VM on TCP socket 44474 -> 10004
$ ./out/DebugIA32Clang/fletch run file main.dart 
Compiled 'file:///usr/local/google/home/mit/dev/github/fletch/fletch/main.dart' to 3445 commands
```

Expected to get a warning that setPin takes (int, bool) but was given (int, int), but did not get any warning. Is there an equivalent of checked mode?

kasperl, 2015-09-18:
The problem is that your `gpio` variable isn't typed and the type-checker doesn't do any type inferencing. Do you get the correct type checks if you change it to the more verbose:

```
GPIO gpio = new PiMemoryMappedGPIO();
```

mit-mit, 2015-09-18:
Ahh, good point! I do indeed get good feedback in that case:

```
main.dart:8:20:
Warning: 'int' is not assignable to 'bool'.
  gpio.setPin(led, 1);
                   ^
package/gpio/gpio.dart:57:8:
Info: This is the method declaration.
  void setPin(int pin, bool value);
       ^^^^^^
Compiled 'file:///usr/local/google/home/mit/dev/github/fletch/fletch/main.dart' to 3445 commands
```

closed 2015-10-02 by peter-ahe-google